pimusicbox/pimusicbox #310: Generate SSH server keys on first boot

dagobert, 2015-08-02:
Installing Pi-Musicbox using your Image causes everyone to have the same SSH server keys. 
This is a unnecessary risk. I understand that it is probably not very important since the box will probably be only use in a LAN but still, it should not be the case.

Possible fix:
1. Remove SSH server keys on building your image
2. Run "dpkg-reconfigure openssh-server" on first boot, the same as you do it for resizing.

PS: Great distribution!

kingosticks, 2015-08-02:
We should already be doing this (note that we do not use openssh-server):
1. https://github.com/woutervanwijk/Pi-MusicBox/blob/develop/makeimage.sh#L165-L166
2. https://github.com/woutervanwijk/Pi-MusicBox/blob/develop/filechanges/opt/musicbox/startup.sh#L143-L153

dagobert, 2015-08-03:
I am sorry. Of cause you are using dropbear.

I was confused about that, because you have additionally openssh-server installed. You might want to change that.

kingosticks, 2015-08-03:
I believe that is distributed with Raspbian. I suppose we might as well remove it at the very end of create_musicbox.sh.

kingosticks, 2015-09-20:
It doesn't hurt anyone to leave this in.

dagobert, 2015-09-21:
The reason to close this issue is not valid, in my opinion. It might not hurt now but it could hurt in the future.
Just imagine that you might switch to openssh-server in the future. Than you have a problem unless you remember this issue.

In total it is bad practice to not deal with this, specially since it is a potential security risk. It is basically no big deal to handle it now but could backfire fatally later.

kingosticks, 2015-09-21:
In switching from dropbear to openssh you'd remove those lines I linked to in startup.sh and struggle to avoid the issue of key generation. 

However, I've been thinking we should move the filesystem resize to a one-time init script and having SSH key regeneration in there, as per one of your original suggestions, seems sensible. It's then also something which we can confine to just the pimusicbox image as people installing the (future) Debian packages onto their existing system wouldn't want it.

We might as well also remove the contents of /etc/ssh in makeimage.sh

dagobert, 2015-09-21:
Thank you for reconsidering my suggestion.

I think, if you have openssh-server installed, as you do even though it is not running, its setup should be secure. This means, as you suggest, either removing the keys on build, or removing them or regenerating them on first boot.

jonathanselea, 2016-09-10:
Sorry for waking this "issue" up again.
But I seems unable to SSH into the Pi?

nerdoc, 2020-11-13:
@wioxjk It seems that this is another issue. Filed it [here](https://github.com/pimusicbox/pimusicbox/issues/552).