pimusicbox/pimusicbox #311: Many files in MusicBox 0.6 image do NOT match files in GitHub

DavisNT, 2015-08-03:
I have compared files in `filechanges` folder of the commit 4d3d304c8c459a130df53baf29a938ea17ea2e56 (which looks closest to the 0.6 release commit) with the actual files from `musicbox0.6.img` and there are many inconsistencies:
 * `etc/init.d/mopidy` differ by contents
 * `etc/modprobe.d/8188eu.conf` differ by contents
 * `etc/modules` differ by contents
 * `etc/mopidy/logging.conf` differ by contents
 * `etc/nsswitch.conf` differ by contents
 * `etc/rc.local` differ by contents
 * `etc/samba/smb.conf` differ by contents
 * `opt/musicbox/startup.sh` differ by contents
 * `etc/monit/conf.d/mopidy` is not a symlink in Git

For many of these files there is no version in Git that matches the file in image.
This is related to #303 and #302 

P.S. I believe that it would be great to have ability to update Pi MusicBox from Git without need for "complete reinstall" on each release (I was writing a script for this, but now it looks like many files are not tracked in Git as they are changed).

kingosticks, 2015-08-03:
Yep, you are 100% correct, it's inconsistent with git. I am working on an updated create script which I'll finish off and commit when I am back from holiday.

DavisNT, 2015-08-07:
Great!
Thank you very much! :) It would be great if you could do the GitFlow style release (commit the released version in `master` branch)...

kingosticks, 2015-08-12:
Even so, it would be good to have a constant history. @woutervanwijk would you be able to commit the extra changes that got into the 0.6 release?

jcass77, 2015-08-14:
I've always used ```create_musicbox.sh``` to quickly rebuild various MusicBox instances that I hack around on, so never even realised that 0.6 was out of sync with the git repository.

It would be interesting to know if the differences are material and affects the functionality in any discernible way.

kingosticks, 2015-08-31:
I've just gone through changes at https://github.com/pimusicbox/pimusicbox/pull/313. I can see that a few are pretty benign but we'd need @woutervanwijk to explain the rest. In particular I'd be interested in:
* `etc/modprobe.d/8188eu.conf`
* `etc/modules`
* `etc/nsswitch.conf`
* `etc/samba/smb.conf`

The `opt/musicbox/startup.sh`change was unfortunate since that's a regression (https://github.com/pimusicbox/pimusicbox/pull/262)

I don't see a difference in `etc/rc.local`.

woutervanwijk, 2015-09-14:
etc/modprobe.d/8188eu.conf: 
added to get the 8188eu chipset (TL-WN725N) working:
https://www.raspberrypi.org/forums/viewtopic.php?t=55779

etc/modules:
snd_bcm2835 is for analog audio

snd_soc_bcm2708_i2s is the i2s interface (for hifiberry and other cards)

bcm2708_dmaengine is for hifiberry too

ipv6 (ipv6 which mopidy need(ed?) )

bcm2708_wdog is the hardware watchdog which resets the pi if it's really hung, has a bad day, fed up with it all, etc.
http://blog.ricardoarturocabral.com/2013/01/auto-reboot-hung-raspberry-pi-using-on.html

jcass77, 2015-12-23:
This has been done right? Should we close this issue?

closed 2015-12-23 by kingosticks