pimusicbox/pimusicbox #312: Move to pimusicbox organisation

kingosticks, 2015-08-12:
This issue tracks the move to a Pi Musicbox GitHub organisation at https://github.com/pimusicbox.

Hopefully the move will allow for greater community involvement but it's important that @woutervanwijk's original vision for Pi Musicbox with *a focus on usability and simplicity* is preserved as it's this we owe for the projects popularity.

* [ ] Commit the released v0.6 files to GitHub (https://github.com/woutervanwijk/Pi-MusicBox/issues/311) 
* [ ] Transfer this repository
* [ ] Transfer the [mopidy-websettings](https://github.com/woutervanwijk/mopidy-websettings)
 and [Mopidy-MusicBox-Webclient
](https://github.com/woutervanwijk/Mopidy-MusicBox-Webclient) repositories
* [ ] Assign Maintainer Roles on PyPi for mopidy-websettings and Mopidy-MusicBox-Webclient
* [ ] Host pimusicbox.com on GitHub Pages (https://github.com/pimusicbox/pimusicbox.github.io/pull/1) 

@woutervanwijk, I think we need you for all of these.  None of this blocks the v1.0 release but it'd be great to get things moving when you can.

kingosticks, 2015-08-12:
* For transferring the repositories: see https://help.github.com/articles/transferring-a-repository/
* For changing roles on PyPi: visit the project page and click 'roles' near the top

jodal, 2015-08-20:
This seems completed, with the exception of #311 which is tracked as its own issue assigned to @woutervanwijk. Closing :-)

closed 2015-08-20 by jodal