pimusicbox/pimusicbox #314: Add a settings.ini entry to downmix to mono

hadess, 2015-09-09:
My "old-timey radio" modernisation hit a snag in that I only have one speaker. Linking both right and left outputs to the speaker resulted in a huge amount of noise. The setting would make it such that right and left channels would be mixed together, and output on both right and left channels.

http://ubuntuforums.org/showthread.php?t=1384860&p=9175251#post9175251 mentions doing that in PulseAudio as:
`load-module module-remap-sink sink_name=mono master=<the name of your main output sink> channels=4 channel_map=left,right,left,right master_channel_map=left,left,right,right`

And in http://superuser.com/questions/155522/force-downmix-to-mono-on-linux for ALSA:
```
pcm.!default makemono

pcm.makemono {
    type route
    slave.pcm "hw:0"
    ttable {
        0.0 1    # in-channel 0, out-channel 0, 100% volume
        1.0 1    # in-channel 1, out-channel 0, 100% volume
    }
}
```

jodal, 2015-09-09:
Ref. https://groups.google.com/d/msg/mopidy/CkkGY7YPnys/U-QUMWqLSO0J, you should be able to have Mopidy/GStreamer downmix to mono by setting Mopidy's `audio/output` config:

``` ini
[audio]
output = capsfilter caps="audio/x-raw-int,channels=1" ! autoaudiosink
```

When the time comes and Mopidy updates to GStreamer 1.x, this will probably be more correct:

``` ini
[audio]
output = capsfilter caps="audio/x-raw,channels=1" ! autoaudiosink
```

hadess, 2015-09-10:
Right, I was hoping for that setting to be in PiMusicBox's configuration instead, as other audio related "static" configurations are done there (such as whether to use hdmi or the jack as a speaker, add-on audio cards, etc.)

kingosticks, 2015-09-10:
It probably would want to set this globally (rather than just for Mopidy) so that shairport output behaves the same.