pimusicbox/pimusicbox #318: Mixer setting gets reverted by hardware mixer

mjtorn, 2015-09-24:
Hi!

I have an external dac, and issues with setting the mixer.

Hardware mixer is off, so I toggle it on, assuming this will enable alsamixer.

Instead I don't get the web ui, and the log contains this:

    2015-09-24 11:00:56,509 - ERROR    Did not find unique mixer "autoaudiomixer". Alternatives are: software, alsamixer
    2015-09-24 11:00:56,513 - ERROR    'int' object is not callable
    Traceback (most recent call last):
      File "/usr/local/lib/python2.7/dist-packages/mopidy/__main__.py", line 127, in main
        return args.command.run(args, proxied_config)
      File "/usr/local/lib/python2.7/dist-packages/mopidy/commands.py", line 264, in run
        mixer_class = self.get_mixer_class(config, args.registry['mixer'])
      File "/usr/local/lib/python2.7/dist-packages/mopidy/commands.py", line 307, in get_mixer_class
        process.exit_process()
      File "/usr/local/lib/python2.7/dist-packages/mopidy/utils/process.py", line 20, in exit_process
        thread.interrupt_main()
    TypeError: 'int' object is not callable

I got to enable alsamixer by actually editing the config manually(!) and reboot.

Looking at the settings UI, hardware mixer appears to be off again, but I guess it's on because the conf still says alsamixer.

Could this be fixed? It would be nice if the web UI reflected the actual configuration :)

Thanks!

PS.
While I was figuring this out, I got this in the log

    2015-09-24 10:13:07,675 - WARNING  Found musicbox_webclient configuration errors, the extension has been automatically disabled:
    2015-09-24 10:13:07,677 - WARNING    musicbox_webclient/musicbox unknown config key.

which led to the web UI not starting up.

I commented it out like this

    [musicbox_webclient]
    enabled = true
    #musicbox = true

Not sure if that's related.

kingosticks, 2015-09-24:
Editing settings.ini manually is perfectly fine, it's well documented to
support this.

There are unfortunately a few bugs with the websettings module in the
current version. It needs updating and I've not gotten around to it yet.
The unknown musicbox key should just be a warning and warnings should not
prevent mopidy starting. If you comment it out you probably won't see the
musicbox specific sections in the Web interface I.e. Settings, system

mjtorn, 2015-09-25:
Is this something addressed post-0.6? I just upgraded using `pip` so should I try to reproduce?

Editing the file is ok and all, but for Joe Q Average it's prohibitive :(