pimusicbox/pimusicbox #319: Scanning through the web UI not found

mjtorn, 2015-09-24:
Hi!

After I got my conf going, I figured I would scan for files to get music playing.

This is almost all my CDs (I think I have fewer than a hundred I haven't ripped yet)

    INFO     Found 9637 files in media_dir.
    INFO     Checking 0 tracks from library.
    INFO     Removing 0 missing tracks.
    INFO     Found 9595 tracks which need to be updated.

This used to take a short eternity on Volumio, so I'm pretty sure I don't want to do this every boot, but only when I buy and rip something new. Unfortunately I couldn't find a button for it in the UI.

Now I'm waiting for `time /etc/init.d/mopidy run local scan` to finish.

kingosticks, 2015-09-24:
The scan should only update the missing/new files so subsequent scans
should be quick.

mjtorn, 2015-09-25:
Let me get back to this once I finish upgrading. Got to ask though, is there any way to trigger the update through the UI?

PS.
Yeah, when I replaced my entire collection with two singles (the other of which failed to be read due to samba and `:` in the directory name) the update was pretty fast.