pimusicbox/pimusicbox #321: Track/file browsing has no directories

mjtorn, 2015-09-24:
After I got all my FLACs up to date on pimusicbox, they're shown as flat files.

Volumio gave me directories a-ok, and I'm kind of puzzled over not having them;
would this be a hard thing to implement?

kingosticks, 2015-09-24:
You have no directory structure at all? I have never heard of that, filesystem directories should be supported - the browsing page would be pretty unusable without them. I'm not sure where to start with this, perhaps you can provide your scan log and mopidy log in a pastebin?

mjtorn, 2015-09-25:
I took a test subset with two Ayreon singles for humane scan times, it also fits in this comment:

    2015-09-25 08:26:52,455 - INFO     Starting Mopidy 0.19.5
    2015-09-25 08:26:55,073 - INFO     Loading config from: builtin defaults, /etc/mopidy/extensions.d/spotify.conf, /etc/mopidy/mopidy.conf, command line options
    2015-09-25 08:26:56,819 - INFO     Enabled extensions: stream, spotify_tunigo, podcast-gpodder, http-kuechenradio, softwaremixer, moped, musicbox_webclient, websettings, tunein, mopify, podcast-itunes, local, mpd, http, dirble, podcast, alsamixer, mobile, local-sqlite, youtube, simple-webclient, local-whoosh
    2015-09-25 08:26:56,826 - INFO     Disabled extensions: spotify, gmusic, scrobbler, subsonic, somafm, internetarchive, audioaddict, soundcloud
    2015-09-25 08:27:15,080 - INFO     Found 6 files in media_dir.
    2015-09-25 08:27:15,109 - INFO     Checking 8222 tracks from library.
    2015-09-25 08:27:34,871 - INFO     Removing 8218 missing tracks.
    2015-09-25 08:28:27,066 - INFO     Found 2 tracks which need to be updated.
    2015-09-25 08:28:27,073 - INFO     Scanning...
    2015-09-25 08:28:28,187 - WARNING  Failed local:track:Network/Ayreon/2000%20-%20Temple%20of%20the%20Cat/musicbrainz.id: The stream is of a different type than handled by this element.
    2015-09-25 08:28:28,272 - WARNING  Failed local:track:Network/Ayreon/2000%20-%20Temple%20of%20the%20Cat/musicbrainz.release: The stream is of a different type than handled by this element.
    2015-09-25 08:28:28,279 - INFO     Scanned 2 of 2 files in 1s.
    2015-09-25 08:28:30,367 - INFO     Done scanning.

Next I hit `/var/lib/mopidy/localdata/sqlite/library.db`

    uri	name	album	artists	composers	performers	genre	track_no	disc_no	date	length	bitrate	commentmusicbrainz_id	last_modified
    local:track:Network/Ayreon/2000%20-%20Temple%20of%20the%20Cat/01.%20Temple%20of%20the%20Cat%20%28single%20version%29.flac	01. Temple of the Cat (single version).flac									196093		1326033105
    local:track:Network/Ayreon/2000%20-%20Temple%20of%20the%20Cat/02.%20Temple%20of%20the%20Cat%20%28Vocals%20Lana%20Lane%29.flac	02. Temple of the Cat (Vocals Lana Lane).flac									241933		1326033164
    local:track:Network/Ayreon/2000%20-%20Temple%20of%20the%20Cat/03.%20My%20House%20on%20Mars.flac	03. My House on Mars.flac	503106				1326033268
    local:track:Network/Ayreon/2000%20-%20Temple%20of%20the%20Cat/04.%20Valley%20of%20the%20Queens.flac	04. Valley of the Queens.flac									146760				1326033296

The path elements are there, will try to look into how Mopidy parses that stuff, but I can confirm I have the tracks off the Temple of the Cat single without any directory structure in the web UI.

Should this be considered a Mopidy - as opposed to MusicBox - problem?

This is 0.6, have there been any changes since this? The build procedure looked a bit hard to tackle just now to test against HEAD.

PS.
Nice Samba moment
`ls: reading directory /music/Network/Ayreon/2004 - Day Eleven: Love/: Not a directory` - I thought having `:` in directory names was fixed already, _grumble_...

kingosticks, 2015-09-25:
Easiest way to work out where the problem lies is to try another webclient
or try an mpd client. Musicbox 0.6 uses an old version of mopidy (0.19.5)
so any issue may even be fixed in the latest version. There are
instructions in the musicbox docs website on how to update but it's a bit
of work and possibly outdated. New version is being tested.

mjtorn, 2015-09-25:
Which version is that? 1.1.1?

Just a quick confirmation: I tried SMPC on my phone, and I can confirm there are no directories with it either. Shows the files with `%20` in them as well, should probably un-urlify those on the way out.

Another webclient is awkward enough that I might as well go back to Volumio, though I'd much rather run something with a clean Python design.

No beta image to try? I'll try the upgrade asap anyway.

Thanks!

mjtorn, 2015-09-25:
`libasound2-dev` was complained about by `pyalsaaudio`, it's missing in https://github.com/pimusicbox/pimusicbox/blob/develop/docs/upgrading.rst

I had to disable `musicbox = true` earlier, and I'm pretty sure the warning went fatal (let's get back to that in #318 if required) so I enabled it manually instead of adding a duplicate section with `echo -e`.

This issue persists; there is no directory structure and looking on my phone files have `%20` in them.

Thanks for the upgrade instructions though :)

So should the web UI or Mopidy be taking care of this?

Not sure if this was in the UI before, but `Files` and `Local Media -> Tracks` are separate. Should the `Files` contain files or what's that for?

Thanks!