pimusicbox/pimusicbox #324: Use debian jessie raspbian

kingosticks, 2015-09-30:
A debian jessie version of raspbian has now been released and we should use this as the basis for the next version of musicbox.

kingosticks, 2015-10-02:
Worth noting in particular that the default boot behaviour is to start a desktop GUI and automatically log in as 'pi' user. This systemd magic can be disabled as seen at https://github.com/raspberrypi/rc_gui/blob/master/raspi-config#L671

dbrgn, 2015-10-03:
Yep, that would be great :)

jcass77, 2015-12-23:
Has any work been done on this yet? Perhaps we can add a separate branch to the official repository so that members of the community can contribute towards this goal?

kingosticks, 2015-12-23:
I've been stabbing at it every so often (for about 6 months!?...wow) and my local branch ended up in a mess. I've (finally) got 3 days of holiday coming up and I'll push the sane bits I find. I've been meaning to do it for ages, I just need prodding -  thanks.

adamcik, 2015-12-25:
A small side note, there seems to also be a lite raspbian prebuilt image for jessie, which might serve as a nice base if not going the debbootstrap route.

jodal, 2015-12-25:
AFAIK, using the minimal prebuilt Raspbian Jessie image is the current plan.

duncangibb, 2017-03-04:
Hi.  I realise this is a dormant issue, but I was wondering if it's still the plan to move to the Jessie Lite image as a base.  I saw a commit on the develop branch a week or so ago switching from pip-installed mopidy and extension components to proper packages (https://github.com/pimusicbox/pimusicbox/commit/fda1bef6a5dea3d9179485cda042104fb98c29d6) but another a few days ago going back to pip because the packages for Wheezy are too old (https://github.com/pimusicbox/pimusicbox/commit/68c2d6b48da85d3eb74f27a333288dd4b224fbb3).

This kind of thing might be easier if the image were based on a current version of the underlying distro.  I would also expect it to be less effort to wrap anything we want to use that's not currently shipped as a .deb in a rudimentary stdeb-produced package if we were targetting Jessie (and Stretch in due course).

I do think there's a lot of merit in shipping PiMusicBox as a plug-and-play system image rather than a metapackage on top of Raspbian.  My ideal of the build process would be a completely automated:

 1) download the raspberrypi.org current official Lite image
 2) chroot into it
 3) purge anything we really don't want/need
 4) hook up some extra apt repos
 5) install all the stuff we need, including our own scripts and default config, as .deb packages
 6) clean up, unmount, compress

Has someone already tried this and discovered I'm being hopelessly naive?

kingosticks, 2017-03-05:
Yes it would be a lot easier but have been many changes in Raspbian since the release of pimusicbox 0.6 and just sticking our existing file changes on top of the latest Raspbian lite image does not produce anything useful. I've did some (maybe all, I don't remember) of the fixes in the 1.0 branch. 

What you've described is the plan, the deb part is described in another issue on here somewhere. It's then trivial to automate it for use with Travis (via Docker and QEMU). With the exception of maybe using pi-gen rather than the actual base Raspbian lite images. I've done some of this in my local branch which I need to push (and ideally cleanup). 

Most recently, I've just been trying to get a new image out quickly to finally support all the new Raspberry Pi versions. And give a bit of breathing space to then continue working on v1.0. The quickest way to do that (without having to test the world) is to just build on the v0.6 image. Sadly,as those commits show, it's a bit of a mess.

duncangibb, 2017-03-05:
Thanks for the swift reply.  I wouldn't expect blasting the 0.6 scripts onto a Jessie image to work; my point was more that basing off Jessie should provide a way to support newer hardware (#257) and get access to a more current set of upstream packages.

I've not used pi-gen but it looks like an elaborate wrapper around debootstrap, which I have.  Are you thinking its "Stage 1" absolutely minimal system would be cleaner and smaller than "Stage 2" which apparently _is_ Raspbian Lite?  Although if 800MB is "Lite" I don't need to worry about my diet ;-)

I did see the "everything should be .deb" issues (#16, #308) and I definitely think that's worth doing - IMLE mixing pip with distro packages frequently ends in tears.  My ulterior motive is that I want to do some integration with mopidy-ircontrol and not mess around managing dependencies by hand.

There are always going to be different priorities (eg personally I don't care about supporting Pi 1, or 1GB SD cards, or Spotify, since I have none of those things) so for many users getting newer versions onto a known-good base is clearly the right thing to do.

I'll take another look at the 1.0 branch an see if I can build anything useful off it.

kingosticks, 2017-03-05:
I think we are on the same page here and have the same goals. I don't think
you can even buy SD cards that small these days!

That branch is as good as dead, it was just a first attempt at migrating
the scripts (and the custom settings, some of which no longer apply/makes
sense). I'm away right now but I will (try again to) push the local stuff I
have when I return.

xavierpeich, 2017-08-25:
I'm also interested in this. I'd like to use my Pi both for MusicBox and RetroPie, but it does not seem possible at the moment because RetroPie requires Jessie. I understand it's not that easy. Is it still on the roadmap?

kingosticks, 2017-08-25:
It's been on hold during 0.7 development but that's coming to an end
(finally). Yes, it's next up, perhaps the next actually image release will
be stretch but the software should be compatible with jessie too.

SoAG, 2017-11-08:
Is there any news on this? Also I'd be happy to help out, though I am still a novice when it comes to building images for RPI.

kingosticks, 2017-11-08:
Well I guess it's outdated as we will obviously leapfrog jessie and go straight to stretch. Is there something in particular you need jessie/stretch for?

adamcik, 2019-07-21:
s/jessie/buster/ by now. This would also allow us to choose between a pure debian and a raspian.