pimusicbox/pimusicbox #327: Documenting Python package versions in each release

DavisNT, 2015-10-11:
I would like to suggest documenting versions of Python packages (installed by PIP) contained in each Pi MusicBox release.

This could be accomplished by adding a simple `pip freeze > path_to_some_file_in_git` to the Pi MusicBox build script.

This feature will make Pi MusicBox more documented and increase reproducibility of building Pi MusicBox images (as well as help users to update Pi MusicBox without erasing SD card).

kingosticks, 2015-10-11:
In the version I'm playing with now, most of the extensions are installed
from apt instead. I think it's only really the webclients left in pip. Can
we achieve the same thing through pinning in apt?

DavisNT, 2015-10-28:
Main goal of this suggestion is to give users ability to update Pi MusicBox without reinstalling (and loosing local media, playlists etc. or struggling with their backup).

P.S. I probably didn't get the idea how pinning could help here...