pimusicbox/pimusicbox #328: Can't access interface anymore

MalcolmMielle, 2015-10-13:
Hey,

I've been using the i music box for a while but I don't seem to be able to access the  Pi MusicBox home screen from my brwoser (Chrome or Firefox) anymore. I don't want could have triggered this, the latest thing I installed was a apache http thing for wordpress.

Every time I try to access the pi music box, I have

```

This webpage is not available

ERR_NAME_NOT_RESOLVED
```

I can access the box fine with `ssh` and from my android phone.

Sorry if that's not where I was supposed to ask this question. Do you have any idea how to find the reason for this or what could be the problem ?

I'm using Fedora23

kingosticks, 2015-10-13:
Musicbox uses ip tables rules to redirect port 80 to mopidy http server
running on port 6680. I think apache probably expects to use port 80 so
perhaps you have some issue there. Disabling the apache service would be a
quick easy test to do. Or look in the apache log files.

kingosticks, 2015-10-13:
Oh, and the musicbox subgroup of http://discuss.mopidy.com is usually the
best place to ask usage questions like this.

MalcolmMielle, 2015-10-13:
Ok I'll move there to answer you :+1:

MalcolmMielle, 2015-10-13:
Moved the topic [here](https://discuss.mopidy.com/t/cant-access-interface-of-pi-music-box-anymore/933)

kingosticks, 2017-03-22:
If this is still an issue in the latest release please reopen.

closed 2017-03-22 by kingosticks