pimusicbox/pimusicbox #331: Hebrew chars support on web client

liri, 2015-10-22:
Hi,
i'm using music box for my retro radio and it's simply great! thank you for you work!

on issue i encountered is the display of song files with hebrew letters, it displays "?" instead. after playing a song it displays the correct hebrew letter on the "now playing" section at the bottom.

will be approciate if fixed on the next release.
Thanks again
Liron
[image]

kingosticks, 2015-10-22:
Hi,

Can you clarify, is this on the browse page of the webclient or the search that it displays? Or playlists? Are these local files or streams/Spotify?

And that's a great looking project, thanks for sharing.

liri, 2015-10-22:
Thank you, glad you like it! i'm attaching some more photos of it.
[image]
[image]
[image]

also, attaching the screenshot of the bug in the browse local files section:
[image]

when playing a file - there are 2 different behaviors:
1) hebrew named file that displays fine when playing it (shot2)
[image]

2) hebrew named file that DOES NOT displays fine when playing it (shot3)
[image]

hope it's clarfiy the problem.
Liron

P.S where should i post a feature request? favorites section will be great to quickly reach my USB folder.