pimusicbox/pimusicbox #332: Spotify Connect Support Opportunity.

artistro08, 2015-11-10:
I'm thinking you can integrate this code on a RPi 2. It'll also work with the chroot method for older RPi's. Someone made a commit that works flawlessly with other programs: https://github.com/Fornoth/spotify-connect-web/pull/18/files

kingosticks, 2015-11-10:
While it's clearly an interesting project and an impressive bit of work I'm
not sure this is something we should include. My initial concerns are
* verifying the contents of the bundled libspotify_embedded_shared.so from
a random 3rd party
* avoiding my Spotify appkey being banned since this is very likely a
violation of their terms of use.

There's also now scope to leverage the Spotify chromecast support which
might be a better avenue to pursue. Are there advantages with connect?

I'm thinking you can integrate this code on a RPi 2. It'll also work with
the chroot method for older RPi's. Someone made a commit that works
flawlessly with other programs:
https://github.com/Fornoth/spotify-connect-web/pull/18/files

woutervanwijk, 2016-04-09:
(from the forum) 
It is now possible to install Connect on Musicbox, using this quick hack:

login using ssh or a keyboard

Issue these commands:

    sudo apt-get install libportaudio2 libvorbisfile3 libavahi-compat-libdnssd1 avahi-daemon -y
    mkdir /opt/librespot
    cd /opt/librespot
    wget http://www.woutervanwijk.nl/mb/connect.zip
    unzip connect.zip

Then add this to the end of /opt/musicbox/startup.sh

    # start Spotify Connect if enabled
    if [ "$INI__spotify__enabled" == "1" ]
    then
        #start with cache to /dev/null to prevent disk full errors
        /opt/librespot/librespot -a /opt/librespot/spotify_appkey.key -b "$INI__spotify__bitrate" -c /dev/null -n "$INI__network__name" -u "$INI__spotify__username" -p "$INI__spotify__password"
    fi

And reboot. Enjoy!

It's not completely stable though, but it works quite well. It will stop spitting out sound if you use shairport or mopidy. A reboot fixes that. See:
https://github.com/plietar/librespot/issues/26

kingosticks, 2017-03-22:
Wheezy uses glibc v2.13 but Rust now requires glibc > v2.13 which stops us from running any recent builds. I tried to build a static binary (using musl) but that fell apart when it came to building static libalsa. If anyone can get that to work then I am all ears. 

Once we move to Jessie this becomes a non-issue.

kingosticks, 2017-03-26:
I was confused, we can still use spotify-web-connect, although it is a bit buggy....

DavidM42, 2017-05-21:
I would also really like spotify-web-connect to be added to Musicbox.
And I can answer you to some of your concerns that this project has been pretty stable for a long time. I am using it for half a year or longer now and my appkey has no problems. It also didn't break in all of that time. Only the volume setting of spotify-web-connect need some adjustment especially on stock raspi analog sound and the mobile app sometimes disconnects from it.
These disconnects are a bit strange because they only happen while using the Android App with spotify-web-connect but not if I control this diy connect speaker from PC or chromecast devices from Android. So only the combo of spotify-web-connect and the Android App (which has many problems and bugs) sometimes produce problems.

moowsje, 2017-06-20:
@woutervanwijk 
I installed Connect for Musicbox as you described in the post above but I cannot find the Pimusicbox using spotify connect after rebooting. Can you maybe help me set this up?

kingosticks, 2017-06-20:
There are good instructions at https://github.com/Fornoth/spotify-connect-web/releases

nightah, 2017-06-23:
An alternative for anybody using an RPi1 would be to download the librespot binary from: https://github.com/herrernst/librespot and following some slightly modified instructions provided by @woutervanwijk above.

At the time of posting the below version is the latest, however, I would suggest ensuring that is the case, if you plan to follow these instructions. Check https://github.com/herrernst/librespot/releases/latest first to confirm.

Via SSH issue the following commands:
```
mkdir /opt/librespot
cd /opt/librespot
wget https://github.com/herrernst/librespot/releases/download/v20170605-39012ba/librespot-linux-armhf-raspberry_pi.zip
unzip librespot-linux-armhf-raspberry_pi.zip
rm librespot-linux-armhf-raspberry_pi.zip
```

Then add this to the end of /opt/musicbox/startup.sh
```
# start Spotify Connect if enabled
if [ "$INI__spotify__enabled" == "1" ]
then
    #start with cache to /dev/null to prevent disk full errors
    /opt/librespot/librespot -b "$INI__spotify__bitrate" -c /dev/null -n "$INI__network__name" -u "$INI__spotify__username" -p "$INI__spotify__password"
fi
```

kingosticks, 2017-06-23:
Does this actually work? Last time I tried to compile the latest version I found that libc was too old when I came to run it. If I'm wrong then that's great news.

nightah, 2017-06-23:
Definitely works, I'm running it on my MusicBox with a RPi1.
Having said that I don't compile it myself manually, I just grab the compiled version from the linked github.

kingosticks, 2017-06-23:
That's great. I'll have to see what steps they do. Does that same binary version work on rpi2/3 do you know? Maybe that's where I came undone before. Hm...

nightah, 2017-06-23:
I've only tried it on my RPi2 and can confirm it works on that too.
According to the page it says it works on all 3 versions and the zero so I would suspect so, however cannot confirm the rest.

kingosticks, 2017-06-23:
Using `-c /dev/null` doesn't work for me: 
```
INFO:librespot: librespot 1a42c31 (2017-03-26). Built on 2017-06-05.
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error { repr: Os { code: 20, message: "Not a directory" } }', /checkout/src/libcore/result.rs:859
note: Run with `RUST_BACKTRACE=1` for a backtrace.
```

nightah, 2017-06-23:
Perhaps, I probably should have tested that before suggesting it.
I actually have created a cache dir and specified that with the option.

kingosticks, 2017-06-23:
Not worry, I'll need to test this quite a lot anyway. Specifying the correct mixer will be some fun and games.

kingosticks, 2017-07-12:
So I have added librespot and when it works, it's great. But the problem with reverse engineered software is that it often breaks and we've been a bit unlucky on that front recently. As of tonight the latest librespot version (https://github.com/pimusicbox/pimusicbox/commit/c10a48274d7a5aaa5c18c91725d761ac888f5b82) stopped working when ever-helpful Spotify decided to block it again. I've no doubt it'll be fixed soon enough (the maintainer is very good) but it looks like it's going to be an ongoing support hassle.

nightah, 2017-07-13:
I can't seem to find anything in the upstream issues about it being blocked again, do you have a source or is it just based on your testing?

I'm still running the older version (v20170605-39012ba) without any problems I figured if spotify decided to block it, it would likely affect all versions?

kingosticks, 2017-07-13:
You'll find recent comments on precious closed issues such as https://github.com/plietar/librespot/issues/190#issuecomment-314911759

It looks like they stopped blocking the old agent string hence why your old version works.

kingosticks, 2017-08-01:
Fixed in https://github.com/pimusicbox/pimusicbox/commit/e8e6f318fffe494401a6bcece595e08743f894c0 and available in RC5

closed 2017-08-01 by kingosticks