pimusicbox/pimusicbox #333: Support non windows OS

Punkratz312, 2015-11-12:
We´ve installed the latest version. The Web-UI is reachable and responses well on all OS´s.
But playing music Spotify only works on our Windows clients.

The Spotify browse window is empty, and "currently playing" isn't displayed on the non windows clients.

I believe very strong, that this is a local problem. But anyway the behaviour is very very strange :disappointed:

kingosticks, 2015-11-12:
What web browsers are you using and is this *only* a Spotify problem or is it with all music sources? I don't understand how it can be anything on our end.

Punkratz312, 2015-11-12:
I´ve testet it on safari, firefox and chome on mac (current versions).
My colleague on Debian Firefox.

Only on his windows machine Firefox 38 its working.

We only tested spotify and webradio

Punkratz312, 2015-11-12:
We downgreaded to 0.5.4. Same issues.

Seems to be an JS problem:

Mopidy.js is using the default calling convention. The default will change in the future. You should explicitly specify which calling convention you use.
gui.js:336 Uncaught TypeError: Cannot read property 'getCurrentTrack' of undefinedupdateStatusTimer @ gui.js:336
mopidy.min.js:5 WebSocket connection to 'ws://172.17.12.107:6680/mopidy/ws/' failed: Error in connection establishment: net::ERR_CONNECTION_TIMED_OUT
mopidy.min.js:5 WebSocket error: Event {isTrusted: true}
gui.js:336 Uncaught TypeError: Cannot read property 'getCurrentTrack' of undefinedupdateStatusTimer @ gui.js:336
mopidy.min.js:5 WebSocket connection to 'ws://172.17.12.107:6680/mopidy/ws/' failed: Error in connection establishment: net::ERR_CONNECTION_TIMED_OUT
mopidy.min.js:5 WebSocket error: Event {isTrusted: true}
2gui.js:336 Uncaught TypeError: Cannot read property 'getCurrentTrack' of undefinedupdateStatusTimer @ gui.js:336
mopidy.min.js:5 WebSocket connection to 'ws://172.17.12.107:6680/mopidy/ws/' failed: Error in connection establishment: net::ERR_CONNECTION_TIMED_OUT
mopidy.min.js:5 WebSocket error: Event {isTrusted: true}
2gui.js:336 Uncaught TypeError: Cannot read property 'getCurrentTrack' of undefinedupdateStatusTimer @ gui.js:336

Punkratz312, 2015-11-12:
Logs looks the same under 6.0

Punkratz312, 2015-11-12:
Another colleague of mine just figured out, that its an port address issue. An request gets fired to an non 8080 port and fails because its not reachable. After he changed the port manually to 8080 it worked. But this issue must occure on the windows machine as well...

kingosticks, 2015-11-12:
What requests are going to port 8080? mopidy's websocket is running on 6680

Punkratz312, 2015-11-12:
Seems to be an network issue, sorry for the hype.

closed 2015-11-12 by Punkratz312