pimusicbox/pimusicbox #336: Mopidy deps says not installed, while pip search does.

JustRamon, 2015-12-12:
I just started using Musicbox, but found out that Youtube isn't working for me. I checked **mopidy debs*, but mopdiy-youtube and others weren't there. Then I executed **pip search mopidy** and pip showed me that most of them *were* installed.

Why isn't this working?

jodal, 2015-12-13:
Can you be more specific and provide exact commands of what you did, what you expected and what happened? For example, what does "checked mopidy debs" mean?

PHiAX, 2016-04-30:
I've got the same problem, and it literally means running the command **mopidy deps** which outputs the installed plugins

kingosticks, 2017-03-22:
As of 0.7.0rc4 `mopidy deps` (not debs) works and mopidy-youtube has been upgraded. Whatever problem there was should be no more.

closed 2017-03-22 by kingosticks