pimusicbox/pimusicbox #338: Ability in GUI to start (full or differential) scanning in the background while playing.

berrydejager, 2015-12-27:
While listening to music, using the Pi MusicBox, i would like to scan my music library (which resides on a netwerk location) from the GUI.

It would be nice to be able to choose between a full scan (clean library and re-scan) and differential (only add/remove changes to the library) scan.

jcass77, 2015-12-27:
This sounds analogous to running ```mopidy local clear;mopidy local scan``` in a terminal window.

Adding this functionality to the GUI should be fairly straight forward (see [relevant code in mopidy-websettings](https://github.com/pimusicbox/mopidy-websettings/blob/1c55d944364775b675827a3a7432974e42459298/mopidy_websettings/__init__.py#L143-L169) and [this in mopidy-musicbox-webclient](https://github.com/pimusicbox/mopidy-musicbox-webclient/blob/24cea6d4beede6e183636da89e13ba2b1710d08a/mopidy_musicbox_webclient/static/js/controls.js#L610-L623)).

As far as I know the standard JSON library storage cannot be refreshed while Mopidy is running (as explained [here](https://docs.mopidy.com/en/latest/ext/local/#updating-the-local-library)), so this change will only make sense if the Pi MusicBox switches to using the [Mopidy Local SQLite](https://pypi.python.org/pypi/Mopidy-Local-SQLite/) extension in the standard distribution.

jehon, 2016-04-17:
What about generating the new JSON in the background, and install it for the next restart?

kingosticks, 2016-05-13:
We will certainly switch to using Mopidy-Local-SQLite. https://github.com/pimusicbox/pimusicbox/issues/371