pimusicbox/pimusicbox #339: Can't disable Spotify.

lech, 2016-01-02:
It seems like no matter what I do in an attempt to disable it, Spotify refuses to stand down and is always enabled despite what options say. It seems like this module refuses to unload or it's completely ignoring the settings all together.

kingosticks, 2016-01-02:
Once you change an option like that you must restart, did you do that? Assuming you did then mopidy-spotify should be disabled; you won't be logged into Spotify, nor will you be able to get any Spotify search results. However, you will still see a Spotify section in the 'browse' page but this is mopidy-spotify-tunigo. Is this what you see?

lech, 2016-01-03:
Yeah, I've rebooted several times and I suspect you're right. It's probably the default Spotify entry in the Browse page. I'd feel better with it hidden upon disabling all Spotify settings so it doesn't bug me out like this.

closed 2016-01-03 by lech