pimusicbox/pimusicbox #341: settings reset

CaptainInler, 2016-01-06:
Hi.
Not sure if I've done something wrong or it is a bug..
I use:
- Pi 2 Model B
  - MusicBox 0.6.0
  - Kernel-Release 4.1.13-v7+
- HifiBerry AMP+
  - aplay -l: `card 1:sndrpihifiberry [snd_rpi_hifiberry_amp], device 0: HifiBerry AMP Hifi tas5713-hifi-0 []`

To get sound out of the AMP, I've modified Settings.ini as followed:
`output = hifiberry_amp`
`mixer_volume = 37`
`mixer = alsamixer`
`card = 1`
`control = Master`
So far so good, sound is coming out of my speaker at a descent volume, even after restarting RPi :v: 

But if I change anything in Settings via the WebUI, settings.ini gets changed after restart. But only one entry:
`mixer = software`
With this Parameter, sound is very very loud, even at volume 1 it's still loud :hear_no_evil:. After resetting "mixer = alsamixer" it all good again.

How can I prevent "mixer =" from being reset after I changed some Settings?
Thanks a lot :sunflower:

jcass77, 2016-01-07:
I've never used the websettings UI, but a quick look at the code seems to indicate that if you don't see an 'Audio output' section then it is probably defaulting the mixer to 'software' behind the scenes. websettings hasn't been updated in a while and I am not sure if it is even fully functional anymore - I get several parsing errors on my machine, so that may be the reason why the setting is not showing up on my setup.

jcass77, 2016-01-07:
Just as an aside: the reason the volume is too loud when using the software mixer is probably due to a line in ``/opt/musicbox/setsound`` that always fixes the **hardware** volume at 96%. Look for ``amixer set -c $CARD "$CTL" 96% unmute > /dev/null 2>&1 || true ``.

PR #309 deals with some of this by defaulting the hardware volume to sensible values for the HiFiBerry equipment