pimusicbox/pimusicbox #342: Connection refused on reboot in 0.6

brimur, 2016-01-09:
Not sure what is happening here. I installed MusicBox 0.6 on my Pi2. Everything works fine. I can stream from spotify. Then if i edit any of the settings and click save settings (reboot) in the browser. The Pi reboots but the browser never reconnects. I get an error message saying connection refused. I cannot connect with the browser on port 80 or the andoid app on 6600.

I can still SSH into the Pi but I dont know what to do to troubleshoot. Any help appreciated.

lech, 2016-01-10:
You can find the logs in ```/var/log/mopidy/``` and quickly check the last few lines that came through using ```cat /var/log/mopidy/mopidy.log``` or watch live with ```tail -f /var/log/mopidy/mopidy.log``` (ctrl+c to stop) if you want to try to catch it in the act. Checking the run logs directly is the best place to start.

It could also boil down to the particular micro SD card you're using with the RPi, if you have spare cards by other manufacturers, give one of those a try with a fresh flash. It's possible the card you're using could be among those that don't pair well with the RPi and as a result corrupt their own file system on boot or shortly after.

brimur, 2016-01-11:
Thanks for the reply lech. I swapped out my 32GB Sandisk EVO sd card for the one that came with the Pi2 and wrote the image to that. It seems more stable in that I can connect after a reboot. I am using with my Spotify Prem account and it will still randomly stop responding when I do an artist or album search. Just an fyi. I will close this since my initial problem is solved.

closed 2016-01-11 by brimur

kingosticks, 2016-01-11:
When the search results are very big they can cause problems for clients - MPD clients in particular. Are you seeing this "not responding" behaviour in the webclient too?  How exactly does it stop responding? Page never loads and you get the spinner graphic? Do you have to restart after this or is it recoverable? If you perform the *exact* same action again does it then work?

brimur, 2016-01-12:
When it happens I get a connection refused error on the webpage as if the server crashed. The android client just acts like the server I am pointing at does not exist, times out. i have to reboot to be able to connect again

kingosticks, 2016-01-12:
In that case I would expect some sort of hint in the log which you'll find at `/var/log/mopidy/mopidy.log` as @lech helpfully points out.

brimur, 2016-01-12:
Yeah I checked that but it didn't show anything, it only seemed to log the page, connection  and image requests. I will check again later and see if there is any more. Thanks