pimusicbox/pimusicbox #343: Possible upgrade pathing feasible?

lech, 2016-01-10:
So I started wondering whether or not having an upgrade approach would be convenient (not to be confused with easy) and wonder whether or not the project could install/update much like the RetroPie-Setup project.

Starting with a fresh Wheezy/Jessie raspbian image, update/upgrade as necessary for the base install. The next step would be to grab Git used to clone the update script which could then be executed to prime the environment necessary to satisfy the requirements needed to get Mopidy up and running, everything from the current /boot configurations to setting up audio and hostname.

Once the system is primed with everything necessary, the primary update script could be run which can then fetch all the necessary current stable Mopidy packages to get users to the familiar Musicbox pages. Then, just fetching a new update script as a release rolls out and re-running it again could be enough without having to flash and prep a whole new image.

Could this be possible or am I overlooking some elements which would make this more difficult than it sounds?

kingosticks, 2016-02-04:
Is this just a wrapper around https://github.com/pimusicbox/pimusicbox/blob/develop/create_musicbox.sh ? I don't see what is new here.

lech, 2016-02-07:
Wow, I should really stop writing out loud while half-asleep. But to answer the question more or less, the answer is yes. 

I can't tell what major differences there are between a base mopidy install when fetching your own packages than what goes into the music box install. Obviously much of it is in the configuration but I'm lost on the specifics here so forgive me.

Take a look at https://github.com/RetroPie/RetroPie-Setup - The setup script goes a bit deep with lots of helper scripts pulling all the pieces together. I'm wondering if something similar level of setup could be achieved for MusicBox which satisfies the criteria for what normally would be release image. It may also give those wanting to dive into a terminal a small safety net when things go wrong (mini diagnostics perhaps?).

Right now, I can only think of the requirements necessary would be maintaining a working catalog of packages necessary to reach a release build. Updating could be atomic for when things like Google Music changes login protocol then just update the git repo and run an update/patch script when a stable update/patch is announced (somehow). It could also probably open the door for applying more compatible and recommended front-end options.

kev4ie, 2017-01-03:
Would it be possible to add push buttons to the gpio pins to go to preset streams and also an i2c output with current stream/song displayed

kingosticks, 2017-01-03:
Yes, we can include https://github.com/9and3r/mopidy-ttsgpio. This suggestion belongs in a separate issue (https://github.com/pimusicbox/pimusicbox/issues/389)

adamcik, 2019-07-21:
This could probably be merged into one of the other similar bugs? #6 in particular