pimusicbox/pimusicbox #344: Poor search results

Flandan, 2016-01-11:
When searching for tracks, via youtube on pimusicbox, many relevant results seem to be omitted. The issue seems to be related to special characters, search results with the character '-' almost never appear. This vastly reduces the usability of the youtube service as music is generally uploaded with the naming convention <Artist> - <Track>.

The puzzling thing is that occasionally a track will appear with the '-' symbol, this is making it difficult to troubleshoot. Am I the only one experiencing this? It's happening with multiple fresh installs of 0.6. I have not seen many similar complaints, and I would expect a lot.

tcurdt, 2016-02-07:
I am playing a track from "Sense Field" on Spotify while writing this, yet searching for "Sense Field" does not show any artists at all. No even when searching for "Sense".

Besides all that it's way too slow to be usable.

kingosticks, 2016-02-07:
Did you try to limit the search to spotify only? Slow backends are mostly out of our control.

kingosticks, 2016-02-07:
And this original issue is specifically regarding youtube results, not Spotify. Please make your own issue and make it in the musicbox-webclient repository.

tcurdt, 2016-02-07:
Thanks, wasn't aware of the other repo. https://github.com/pimusicbox/mopidy-musicbox-webclient/issues/160

kingosticks, 2017-03-22:
Mopidy-youtube was upgraded in 0.7.0rc4. Please reopen if this issue is not fixed. If you can provide the exact steps to reproduce the problem that would be very helpful.

closed 2017-03-22 by kingosticks

Flandan, 2017-03-24:
I am currently away from my pi but I will check out the new release when I can and see if there are improvements. Having used the youtube api (which I believe is what is being used in the background here) in an unrelated ruby based project and getting similar issues with the results (although subjectively less pronounced) I am now inclined to believe thats where the issue is. Possibly an issue in how searches are defined?

I will try to look into it myself when I have the time.