pimusicbox/pimusicbox #345: No security

IRTermite, 2016-01-15:
I know the docs say not to put it outside a firewall and that it isn't secure. Are there plans to add Users/Groups? At simplest a simple Admin/Guest auth method? Changing the root password through the interface is nice and all, but when my visitors can also hop on and do the same thing without validation, that's a problem. What would it take to add an Admin and a Guest user to protect and hide certain admin-type functions?

kingosticks, 2016-01-15:
No plans to add users /groups. But it should be simple enough to put mopidy behind nginx (or something) and protect the websettings page with a simple password. That's something I'd like to integrate.

IRTermite, 2016-01-16:
That works. As long as "admin-level options" are behind a password, that is better than nothing. I was actually looking for an .htaccess of sorts or a path to add one on my pi, but could figure out the structure at the time. Now that I found this repo, guess it will be easier to look through.

LaKing, 2016-03-17:
+1 for https and password protection