pimusicbox/pimusicbox #346: Bug: Crashing when spotify continue

clausfuglsang, 2016-01-22:
After I stop playing music from spotify, which basically works fine, the Musicbox will crash if I leave it for more than an hour or so, and then wants to continue playing music again.
If i stop playing, and I get back to it an hour later, I need to reboot it manually if I want to play from spotify again. I only use spotify, so i am not aware of if the problem is the same with the other services.

I've had this issue for 1½ years now.
I'm running the latest release of Musicbox.
Great work though, love the features.

jcass77, 2016-01-22:
Could you perhaps take a look at ```/var/log/mopidy/mopidy.log``` and post any relevant error messages contained therein?

Does it work if you do ```service mopidy restart``` instead of a full reboot of the Musicbox?

kingosticks, 2016-01-22:
And, just to clarify, when you say 'stop' do you mean pause?

lukyluke112, 2016-02-01:
Have the same issue

lukyluke112, 2016-02-01:
And with me it just won't play anything

kingosticks, 2016-02-01:
If you don't your logs there is literally nothing to go on.

lukyluke112, 2016-02-01:
Bit of a noob looked for log file do you use a different name for the log file

B3nihana, 2016-04-18:
I have the same issue also, as I use MusicBox infrequently but solely for Spotify. I am still able to restart MB via the web UI but its very slow to respond (UI very sluggish to load menus, will play songs like there is no bandwidth).

I'll be sure to get the log next time this happens, I'm just so used to rebooting it now!

Original Pi B+ btw

B3nihana, 2016-05-11:
Ok, so this issue happened again and I remember to grab the log.

Basically I start to play a spotify playlist, then pause it after several hours and try to unpause it the next day and nothing happens. I can navigate the GUI but searching for music or trying to start a new playlist results in a spinning busy icon. Sometimes it will start working 30 minutes later, but usually I simply restart and it works.

Hope this helps:
Monit.log
`[UTC` Apr 17 13:08:21] info     : 'mopidy' trying to restart
[UTC Apr 17 13:08:21] info     : 'mopidy' stop: /etc/init.d/mopidy
[UTC Apr 17 13:08:51] info     : 'mopidy' start: /etc/init.d/mopidy
[UTC Apr 17 13:10:51] info     : 'mopidy' connection succeeded to INET[localhost:6680] via TCP
[UTC Apr 17 16:06:38] info     : monit daemon with pid [2478] killed
[UTC Apr 17 16:06:38] info     : 'system_MusicBox' Monit stopped
[UTC Apr 17 16:07:15] info     : Starting monit daemon
[UTC Apr 17 16:07:15] info     : Monit start delay set -- pause for 300s
[UTC Apr 17 16:12:15] info     : 'system_MusicBox' Monit started
[UTC Apr 18 12:05:40] info     : monit daemon with pid [2480] killed
[UTC Apr 18 12:05:40] info     : 'system_MusicBox' Monit stopped
[UTC Apr 18 12:06:42] info     : Starting monit daemon
[UTC Apr 18 12:06:42] info     : Monit start delay set -- pause for 300s
[UTC Apr 18 12:11:42] info     : 'system_MusicBox' Monit started
[UTC Apr 24 08:07:17] info     : monit daemon with pid [2488] killed
[UTC Apr 24 08:07:17] info     : 'system_MusicBox' Monit stopped
[UTC Apr 24 08:07:55] info     : Starting monit daemon
[UTC Apr 24 08:07:55] info     : Monit start delay set -- pause for 300s
[UTC Apr 24 08:12:55] info     : 'system_MusicBox' Monit started
[UTC Apr 25 11:45:46] error    : 'upmpdcli' process is not running
[UTC Apr 25 11:45:46] info     : 'upmpdcli' trying to restart
[UTC Apr 25 11:45:46] info     : 'upmpdcli' start: /etc/init.d/upmpdcli
[UTC Apr 25 11:47:46] info     : 'upmpdcli' process is running with pid 3066
[UTC Apr 29 17:48:54] error    : 'upmpdcli' process is not running
[UTC Apr 29 17:48:54] info     : 'upmpdcli' trying to restart
[UTC Apr 29 17:48:54] info     : 'upmpdcli' start: /etc/init.d/upmpdcli
[UTC Apr 29 17:50:54] info     : 'upmpdcli' process is running with pid 5317
[UTC May  4 05:17:30] info     : Starting monit daemon
[UTC May  4 05:17:30] info     : Monit start delay set -- pause for 300s
[UTC May  4 05:22:30] info     : 'system_MusicBox' Monit started
[UTC May 10 07:36:18] error    : 'mopidy' process is not running
[UTC May 10 07:36:18] info     : 'mopidy' trying to restart
[UTC May 10 07:36:18] info     : 'mopidy' start: /etc/init.d/mopidy
[UTC May 10 07:38:18] info     : 'mopidy' process is running with pid 3691
[UTC May 11 06:27:07] error    : 'mopidy' failed protocol test [HTTP] at INET[localhost:6680] via TCP -- HTTP: Error receiving data -- Resource temporarily unavailable
[UTC May 11 06:29:12] error    : 'mopidy' failed protocol test [HTTP] at INET[localhost:6680] via TCP -- HTTP: Error receiving data -- Resource temporarily unavailable
[UTC May 11 06:29:12] info     : 'mopidy' trying to restart
[UTC May 11 06:29:12] info     : 'mopidy' stop: /etc/init.d/mopidy
[UTC May 11 06:29:42] info     : 'mopidy' start: /etc/init.d/mopidy
[UTC May 11 06:31:43] info     : 'mopidy' connection succeeded to INET[localhost:6680] via TCP`

Mopidy.log

`2016-05-11 06:29:46,473 - INFO     Starting Mopidy 0.19.5
2016-05-11 06:29:48,592 - INFO     Loading config from: builtin defaults, /etc/mopidy/extensions.d/spotify.conf, /etc/mopidy/mopidy.conf, command line options
2016-05-11 06:29:50,262 - INFO     Enabled extensions: mopify, spotify, mpd, http, stream, spotify_tunigo, mobile, podcast-gpodder, local-sqlite, youtube, podcast-itunes, http-kuechenradio, scrobbler, simple-webclient, softwaremixer, local-whoosh, dirble, moped, websettings, podcast, tunein
2016-05-11 06:29:50,265 - INFO     Disabled extensions: alsamixer, gmusic, somafm, subsonic, internetarchive, audioaddict, musicbox_webclient, local, soundcloud
2016-05-11 06:29:50,268 - WARNING  Found alsamixer configuration errors, the extension has been automatically disabled:
2016-05-11 06:29:50,271 - WARNING    alsamixer/control must be set.
2016-05-11 06:29:50,274 - WARNING    alsamixer/card must be set.
2016-05-11 06:29:50,276 - WARNING  Found musicbox_webclient configuration errors, the extension has been automatically disabled:
2016-05-11 06:29:50,279 - WARNING    musicbox_webclient/musicbox unknown config key.
2016-05-11 06:29:50,282 - WARNING  Please fix the extension configuration errors or disable the extensions to silence these messages.
2016-05-11 06:30:07,956 - INFO     Starting Mopidy mixer: SoftwareMixer
2016-05-11 06:30:07,963 - INFO     Mixing using GStreamer software mixing
2016-05-11 06:30:07,976 - INFO     Mixer volume set to 85
2016-05-11 06:30:07,980 - INFO     Starting Mopidy audio
2016-05-11 06:30:07,997 - INFO     Starting Mopidy backends: SpotifyTunigoBackend, TuneInBackend, StreamBackend, PodcastBackend, YoutubeBackend, SpotifyBackend, DirbleBackend
2016-05-11 06:30:08,150 - INFO     Audio output set to "alsasink"
2016-05-11 06:30:08,299 - INFO     Starting Mopidy-Podcast directories: iTunesDirectory, FeedsDirectory, GPodderDirectory
2016-05-11 06:30:08,333 - INFO     Refreshing Mopidy-Podcast directories
2016-05-11 06:30:08,412 - INFO     Starting new HTTP connection (1): itunes.apple.com
2016-05-11 06:30:08,516 - INFO     Mopidy uses SPOTIFY(R) CORE
[... cut ...]
2016-05-11 07:55:46,135 - INFO     200 GET / (127.0.0.1) 15.07ms
2016-05-11 07:57:23,657 - INFO     304 GET /images/icons/play_alt_32x32.png (192.168.1.34) 43.15ms
2016-05-11 07:57:23,766 - INFO     304 GET /images/default_cover.png (192.168.1.34) 69.07ms
2016-05-11 07:57:46,201 - INFO     200 GET / (127.0.0.1) 17.33ms
2016-05-11 07:59:46,346 - INFO     200 GET / (127.0.0.1) 55.81ms
2016-05-11 08:01:46,429 - INFO     200 GET / (127.0.0.1) 22.53ms
2016-05-11 08:03:46,552 - INFO     200 GET / (127.0.0.1) 38.37ms`

kingosticks, 2016-05-12:
Thanks for providing your log unfortunately the LOG_LEVEL is too high to show anything interesting. The problem with increasing the verbosity is that over time it's a great way to wear out your SD card so it's not advisable in the long-term. However, if you have a way to reliably recreate this problem then you could try temporarily setting `level = DEBUG` in `/etc/mopidy/mopidy.log`.

When Mopidy crashes, depending on your browser, you may still be able to navigate the web UI since it's cached by your browser; but anything action that requires new response data will fail and you'll get the spinning icon. 

Can you just clarify when you paused and unpaused, particularly in relation to the events in monit.log.