pimusicbox/pimusicbox #348: shuffle play

tcurdt, 2016-02-07:
Is there a way to shuffle play a playlist?
I cannot find it.

kingosticks, 2016-02-07:
Shuffle the playlist s the leftmost icon on the now playing bar. I think it's up and down arrow. Random play is also down there somewhere. They should all have mouse-over descriptions.

closed 2016-02-07 by kingosticks

kingosticks, 2016-02-07:
Sorry! I mean right-most icon!

tcurdt, 2016-02-07:
Found it. Odd. That one is obvious.
I could have sworn I didn't see it on the phone.

kingosticks, 2016-02-07:
It should be there for all devices, I think it gets a bit squashed down there on small screen devices so we could look at moving them elsewhere. Please create an issue in the webclient repo if you do find a problem on your device.

tcurdt, 2016-02-07:
Will double check, thanks!