pimusicbox/pimusicbox #349: Remove Mopidy-Subsonic

kingosticks, 2016-02-08:
Mopidy-Subsonic appears to be unmaintained and if so, it's not feasible to include it in future pimusicbox releases (https://github.com/pimusicbox/pimusicbox/issues/308). 

I've asked the current maintainer to confirm the status of the project as https://github.com/rattboi/mopidy-subsonic/issues/17

kingosticks, 2016-05-13:
No reply, it'll be removed from the next version. If it seems future development and there is any demand we can add it back.

benjbubu, 2016-06-13:
It's a shame i was using pimusicbox with my subsonic server....furhtermore, i was browsing this repository because my pimusicbox can't boot anymore due to a fatal python error since i upgrade my version of subsonic at 6.0...

novellahub, 2016-06-20:
I have the pimusicbox working with the latest Subsonic 6.0.  The fatal python error has to do with no playlists defined.  Just create one playlist in Subsonic and it will work.

benjbubu, 2016-06-25:
I confirm novellahub's post. Creating a new playlist remove the fatal error at startup. Nice job :)

zmrow, 2016-07-23:
Any idea what the problem is with the mopidy-subsonic project?  Is the maintainer quitting?  Wonder if some of us could pick it up.

kingosticks, 2016-07-25:
I would suggest asking at https://github.com/rattboi/mopidy-subsonic/issues/17 but I don't think we've seen anything from @rattboi in many months now

jolny, 2017-05-19:
Hello guys!
I suggest you include this plugin for Subsonic instead of mopidy-subsonic: https://github.com/Prior99/mopidy-subidy
It works great for Subsonic 6.2 and is actively mantained. I've created a pull request for support for older versions of Subsonic and hhm0 has added a lot of functionality which has yet to be merged into main.

rattboi, 2017-05-19:
Hi all,

I concur that going with another plugin is the right direction. I haven't
worked on mopidy-subsonic in a long time, and in fact have stopped using
both mopidy and subsonic, so maintaining it just isn't realistic anymore :(

Sorry for the radio silence for so long.

kingosticks, 2017-05-20:
OK sounds good to me. Thanks for all your work, @rattboi.

jolny, 2017-05-20:
I've tried using loldaves fork and even adding some later commits by Prior99 and others, but I couldn't get it working. I e-mailed him and he said to try Mopidy-subidy instead, and that worked great!
@rattboi Maybe Mopidy-subsonic should be removed from PyPI? Or possibly if you add another maintainer that can update it. :)

kingosticks, 2017-05-22:
Unfortunately mopidy-subidy [requires Mopidy v2.0+](https://github.com/Prior99/mopidy-subidy/blob/master/setup.py#L26) but even the latest pimusicbox release candidate only provides v1.1.2. I don't know if it actually *needs* this version of Mopidy's API (few things do) but that's what is currently required to install. I appreciate that developing for and maintaining compatibility with older software is undesirable but sadly I cannot include mopidy-subidy in pimusicbox v0.7.