pimusicbox/pimusicbox #350: No License

JoeIaquinto, 2016-02-08:
Does this project have an open source license?
If you do, could it be to the README or a LICENSE.txt file?

jodal, 2016-02-08:
That's a very good question. Pi Musicbox itself is just a set of configuration and scripts to setup all the components to work together, but I agree that the repo should have a license.

Looking at https://github.com/pimusicbox/pimusicbox/graphs/contributors I think that at least @woutervanwijk, @kingosticks, @jcass77 and myself should agree on a license before adding it to the repo. The remaining contributions are (arguably) few/small enough to not track down the contributors and ask about their opinion. If anyone strongly disagrees with this, please speak up.

My suggestion is to go with a permissive license. Either Apache 2.0, like Mopidy, Mopidy-Musicbox-Webclient and Mopidy-WebSettings, or MIT, like some other Mopidy extensions. According to https://www.blackducksoftware.com/resources/data/top-20-open-source-licenses these have 16% and 25% market-/mindshare, while the next permissive license, BSD, is down at 6% and exists in several different variants, which adds to the confusion.

I also suggest that we add the same license to the pimusicbox.github.io repo, once we decide on a license.

kingosticks, 2016-02-08:
I agree with everything @jodal has said above and my preference is Apache 2.0.  

I admit I am unclear what happens when you start trying to distribute a collection of packages with all different licenses (including things that are non-free) all together. Does this collection then become a different licensing problem?

jodal, 2016-02-08:
Pi Musicbox is basically a Linux distribution. The components have their own licenses, and we have a license for the wiring that makes all the components into Pi Musicbox. There's no spread of e.g. copyleft licenses as we're not linking together any libraries of varying licenses.

jcass77, 2016-02-09:
+1 on going for Apache 2.0

woutervanwijk, 2016-02-10:
Same with me, Apache 2.0 is ok

commit 0b52d111 (2016-02-10): Add Apache 2.0 license

Fixes #350
closed 2016-02-10 by jodal