pimusicbox/pimusicbox #351: Search results

grawlybean, 2016-02-18:
Hello,

Version 0.6 has been working beautifully for me for months and I have used the web client primarily to stream using Spotify.
However recently the search facility has stopped working and returns no results when searching for Spotify tracks.
Playback via Spotify playlists works fine and all other functions.
I have looked in the Mopidy log file and no luck. Reboot also does not change things.

Thanks for any help.

jcass77, 2016-02-18:
This has been fixed in Mopidy-Spotify 2.3.1 (see: https://github.com/mopidy/mopidy-spotify/issues/89).

grawlybean, 2016-02-18:
Ok thanks.
I have been unable to update the mopidy-spotify package successfully and now get "unable to locate package ..." when I attempt apt-get install mopidy-spotify.

kingosticks, 2016-02-18:
Thanks to Spotify's complete disregard for their customers (see https://jodal.no/2016/02/18/guide-to-poor-api-management/) the *Spotify* search is completely broken unless you update. 

Unfortunately... updating musicbox is a dark art and not to be undertaken lightly. I wish I could give you a set of working instructions but doing so would require time that's better invested working on a new pimusicbox release. 

But I can tell you your error above is because pimusicbox v0.6 installs Mopidy + extensions exclusively from pip rather than APT. You can try something like this:
```
apt-get -y install build-essential python-dev libffi-dev
service monit stop
service mopidy stop
pip install requests[security]
pip install --upgrade pip
pip install mopidy==1.1.2
pip install mopidy-spotify==2.3.1
service mopidy restart
service monit start
```

Hopefully this should *not update to the very latest version* of Mopidy (v2.0.0) as this requires new gstreamer libraries which I don't think are readily available for pimusicbox v0.6. Mopidy v.1.1.2 should be fine for the time being.

grawlybean, 2016-02-18:
Excellent that fixed things.
Thanks very much for the reply and all your support on this fantastic project.
I love my little Pi Musicbox player and this is now back to full strength.

kingosticks, 2016-02-18:
Brilliant.  Thanks for reporting back that this worked and for the kind words. Hopefully others will also be able to make use of this temporary fix.

kingosticks, 2016-02-18:
Just to add, there is a chance that the pimusicbox startup script may have some problems with the updated version of Mopidy. Something to be aware of and if you do have issues when rebooting then please feel free to reopen this issue and hopefully we can sort it.

closed 2016-02-18 by kingosticks