pimusicbox/pimusicbox #356: No sound through USB

ChrisByrne112, 2016-02-21:
I'm sure this is not a bug, but user error. I'm happy to get my hands dirty and debug this myself, but a few pointers would be greatly appreciated. 

I've a Cambridge Audio Dacmagic 100 and can't get any sound when it's connected to the PI. Analogue audio works perfectly. 

Any ideas for debugging? Thanks in advance.

kingosticks, 2016-02-21:
Detecting and setting the audio device is handled by [setsound.sh](https://github.com/pimusicbox/pimusicbox/blob/develop/filechanges/opt/musicbox/setsound.sh). Presumably this is a USB device and you've already set `output = usb` in `settings.ini`. setsound.sh uses the output of `aplay -l` to try and work out what is connected and it should print out what it finds as it goes along. If it cannot work out what a particular device is it'll assume it's a USB soundcard.

You can just run setsound.sh from the shell rather than rebooting. 

It then sets the system's default sound device by writing `/etc/asound.conf` so check the device there matches your Dacmagic. 

Tthere is a `downsample_usb` option you can try for soundcards that fussy about being told the sample rate. 

Other than that, use `alsamixer` do double check the volume isn't muted or something crazy.

ChrisByrne112, 2016-02-21:
Perfect, thanks for the quick response. I'll check those out this evening and get back to you.

ChrisByrne112, 2016-02-22:
All working now thanks, the card just needed to be un-muted in alsamixer :-)

Where's the best place to automate this, so I don't need to ssh in and un-mute after every reboot?

kingosticks, 2016-02-22:
I think https://github.com/pimusicbox/pimusicbox/blob/develop/filechanges/opt/musicbox/setsound.sh#L245 is trying to do exactly that. Perhaps it doesn't have the correct `CTL` (control) for your particular card. If you go into alsamixer it should list your soundcards controls.

ChrisByrne112, 2016-02-22:
Looks like that script is doing it's job. But the Dacmagic seems to appear as 2 devices. 
```
root@MusicBox:/etc# amixer
Simple mixer control 'Cambridge Audio DAC100 USB 1 Audio In Playb',0
  Capabilities: pswitch penum
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
Simple mixer control 'Cambridge Audio DAC100 USB 1 Audio In Playb',1
  Capabilities: pswitch pswitch-joined penum
  Playback channels: Mono
  Mono: Playback [off]
```
The sound is being routed to the second mono device, not the first stereo device. Do you know how I can force the use of the first stereo device?

ChrisByrne112, 2016-02-22:
OK, I think i'm sorted, adding these 2 lines 

amixer set -c 1 "Cambridge Audio DAC100 USB 1 Audio In Playb"  96% unmute
amixer set -c 1 "Cambridge Audio DAC100 USB 1 Audio In Playb",1  96% mute

to setsound.sh works, I guess they could be added to the CTL list as well, but an edge use case. 

Thanks for the help.

ChrisByrne112, 2016-02-22:
Updated setsound.sh. 
```
for CTL in \
        Master \
        PCM \
        Line \
        "PCM,1" \
        Wave \
        Music \
        AC97 \
        "Master Digital" \
        DAC \
        "DAC,0" \
        "DAC,1" \
        Speaker \
    Playback \
    Digital \
    Aux \
    Front \
    Center \
    "Cambridge Audio DAC100 USB 1 Audio In Playb,0" \
    "Cambridge Audio DAC100 USB 1 Audio In Playb,1"
do
    # Set initial hardware volume
    amixer set -c $CARD "$CTL" 96% unmute > /dev/null 2>&1 || true 
```

closed 2016-02-22 by ChrisByrne112

kingosticks, 2016-02-22:
We need a more generic solution for this else we'll be here until the end of time adding these. I wonder if there is any reason we can't do something like:
```
for x in `amixer controls  | grep layback` ; do amixer set -c $CARD "${x}" 96% unmute ; done
```

ChrisByrne112, 2016-02-22:
Yeah, I think that parsing the output of `amixer controls` would work, so in this case

```
numid=2,iface=MIXER,name='Cambridge Audio DAC100 USB 1 Audio In Playb'
numid=3,iface=MIXER,name='Cambridge Audio DAC100 USB 1 Audio In Playb',index=1
numid=1,iface=PCM,name='Playback Channel Map'
```

would need to produce

```
Cambridge Audio DAC100 USB 1 Audio In Playb,0
Cambridge Audio DAC100 USB 1 Audio In Playb,1
```

Any RegEx experts out there?