pimusicbox/pimusicbox #357: Pi 3

lukyluke112, 2016-02-29:
So I just bought a pi 3 and it will come in a couple of days and I would like to know if pi music box will work out of the box with the pi 3

kingosticks, 2016-02-29:
> You’ll need a recent NOOBS or Raspbian image from our downloads page

_(source: https://www.raspberrypi.org/blog/raspberry-pi-3-on-sale)_

Since Musicbox v0.6 is built on a really old image I doubt it'll boot. Usually it's a case of copying over out boot.elf and fixup.dat files but I really can't be sure until mine turns up (also in a few days, I hope). 

This is yet another great reason why we really need a new release but time has been against me lately.

lukyluke112, 2016-02-29:
Thanks I'll try that when I get mine and see if it works

Subsack, 2016-02-29:
Hi,

I can confirm that it doesn't boot out of the box, however copying boot.elf and fixup.dat with respective _cd files did make it boot as kingosticks thought, thanks!

I haven't got around to do much testing yet but it seems like the on-board WiFi will need some attention.

kingosticks, 2016-03-02:
Thanks @sdbakker . The onboard wifi is reportedly the same as the official wifi dongle (the BCM43143?) which uses the brcmfmac driver. [Wheezy has firmware-brcm80211 v0.36](https://packages.debian.org/wheezy/firmware-brcm80211) which doesn't appear to contain the necessary firmware files ([Jessie has v0.43](https://packages.debian.org/jessie/firmware-brcm80211) and definitely does).

lukyluke112, 2016-03-02:
but if I want to update the raspian would i reck my musicbox?

kingosticks, 2016-03-02:
Yes. I'm still working on a new jessie based version. You should be also able to copy the extra firmware files over if you want to use the onboard wifi. Or you might even get away with directly installing the [deb from wheezy-backports](https://packages.debian.org/wheezy-backports/all/firmware-brcm80211/download) since it's not architecture dependent...

lukyluke112, 2016-03-02:
Cool ill try that thanks

davidpronk, 2016-03-03:
> I can confirm that it doesn't boot out of the box, however copying boot.elf and fixup.dat with respective _cd files did make it boot as kingosticks thought, thanks!

@kingosticks can you specify exactly what files you copied. I'm not sure which ones to include.

kingosticks, 2016-03-04:
I think you'd be better off directing that question at @Subsack but I would guess:
* `*.dat`
* `*.elf`

would be sufficient. Maybe `bootcode.bin` too, but I'm not sure if that's actually needed.

davidpronk, 2016-03-04:
My bad... I meant to do so but copied the name from the wrong post.

davidpronk, 2016-03-04:
:+1: 
By overwriting all *.dat and *.elf files from `2016-02-26-raspbian-jessie-lite.img` to Volumio2 2.0.0-alpha4 I got the Pi 3 up and running Volumio.

fumblingwithit, 2016-03-12:
Got MusicBox to rock like this:
mounting jessie-lite gives two mount points `boot` and `1.3G` 
mounting musicbox-0.6 gives two mount points `MUSICBOX` and `8.0G`

The two secondary mount points will probably be called something else for you
For the sake of argument I'll call them `Jessie1.3` and `MusicBox8.0`

Copy files....
`cp -r boot/ MUSICBOX/`

`cp -r [Jessie1.3]/lib/modules [MusicBox8.0]/`

Update apt-repositories from wheezy to jessie
`apt-get update && apt-get upgrade`
`reboot`

There were a few differences of some scripts upon upgrading, but I choose "keep maintainer" in most cases. I kept current version for watchdog

Update firmware
`apt-get install rpi-update`
`rpi-update`
`reboot`

I have yet to get wifi working....

fumblingwithit, 2016-03-14:
A quick showdown of my config in regards to wifi:
```
root@MusicBox:~# cat /etc/network/interfaces                                           
auto lo
iface lo inet loopback

#auto eth0
allow-hotplug eth0
iface eth0 inet dhcp

auto wlan0
allow-hotplug wlan0
iface wlan0 inet dhcp

wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

wireless-power off
iface default inet dhcp
```
Remove a nag about wireless p80211 driver that keeps my wireless from working:
```
apt-get remove linux-wlan-ng
```

My configuration of the wpa_supplicant.conf:
```
root@MusicBox:~# cat /etc/wpa_supplicant/wpa_supplicant.conf 
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="MY_SSID"
    psk="MY_WPA_KEY"
}
```
At current I get wireless access through to my MusicBox, although the throughput is terrible. I guess the drivers for rpi3 have yet to mature.

Hope alle this helps others, cheers all.

bonelifer, 2016-03-14:
There's something about turning off power management to fix the RPi3's wifi for the time being. I believe theres a fix upstream, but currently would require using rpi-update.  Here's one topic showing a command to fix this for now: https://www.raspberrypi.org/forums/viewtopic.php?p=928480#p928480

fumblingwithit, 2016-03-15:
@bonelifer - It appears to have helped to do an ``` apt-get update && apt-get upgrade``` today. The article you mention basically tells you to insert ```wireless-power off``` into your interfaces and upgrade to the latest firmware. - Thx for the reminder

lukyluke112, 2016-04-07:
I tried to get the .elf and .dat files but flashing it via noobs and I flashed raspbian a couple of times. I could not find any .dat files and only 1 .elf file

fumblingwithit, 2016-04-14:
@lukyluke112 sound like you do not have the correct folder mounted

vantezzen, 2016-04-21:
If someone still can't get it work, here's a tutorial: [http://bennetthollstein.de/wordpress/?p=147](http://bennetthollstein.de/wordpress/?p=147)

rnamdos, 2016-05-08:
Just here to say that I followed @fumblingwithit instructions and I got my rpi3 to stream over my iPhone 4G hotspot. Every other attempt failed miserably at firmware update.

You might also need to also install the Broadcom driver. Follow this (copied from [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138629))
_
# Remove the Driver
modprobe -r brcmfmac
# Load in the Driver
modprobe brcmfmac
# Check Dmesg for any errors
dmesg
# List all network ports
ifconfig -a

If you see an error in the dmesg log like this, then your likley missing the needed firmware files
brcmfmac_sdio mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2

In that case go to https://github.com/RPi-Distro/firmware- ... 80211/brcm
and download the brcmfmac43430-sdio.bin and brcmfmac43430-sdio.txt files into the directory
/lib/firmware/brcm/ then try again_

rgov, 2016-08-17:
What is the latest release of Raspbian that you have successfully installed? I tried the above with `2016-05-27-raspbian-jessie-lite.img` without success. I'll try with `2016-02-26` since that's the last version explicitly mentioned above.

rgov, 2016-08-17:
This did **not** work to get from Wheezy to Jessie (which is perhaps outside the scope of this bug):

1. Edit `/etc/apt/sources.list` and all of the files in `/etc/apt/sources.list.d` to replace `wheezy` with `jessie`. For `mopidy.list`, execute `wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/jessie.list`.

2. Run `apt-get update && apt-get dist-upgrade`. This takes a while.

3. Choose "no" when prompted if you want to disable SSH as the root user. Choose "yes" when prompted if you want Apt to automatically restart services after upgrading their dependencies.

4. Whenever asked if you want to replace a configuration file, say "yes". The original Pi MusicBox configuration files are [here](https://github.com/pimusicbox/pimusicbox/tree/develop/filechanges/etc) for future reference.

5. Reboot the Raspberry Pi. Did not come back up.

shawaj, 2016-08-21:
@kingosticks can you send me an email to shawaj at gmail - got something to discuss with you if that's OK?

frob, 2016-09-08:
Curios if it is possible to just run the install script on the latest raspian on a Pi3 instead of trying to upgrade.

stevanguily, 2016-12-30:
Hello is what it would be possible to have all the necessary files in a file compiled
Because I start in raspberry and I am a little feneant

PS: I'm French so small mistakes are normal

shanehull, 2017-03-02:
Any update on Pi 3 support?

kingosticks, 2017-03-02:
Yes! v0.7 (support for Pi 3, Zero and ZeroW) is very nearly ready. A few last things to fix up (wireless being one) and it's good to go. If you are really keen you can test the beta at https://drive.google.com/open?id=0ByrmzcQJL5t1c1V0NzhheWl6Tlk

shanehull, 2017-03-02:
Wow, that was quick! Thank you! I'll give it a test now

rgov, 2017-03-02:
I gave up on trying to use PiMusicbox, but it was quite simple to install Mopidy.  I can't give any tech support for this but you can try to follow along.

https://gist.github.com/rgov/ae3a2ceb3bde964b8836d83c88b11924

kingosticks, 2017-03-02:
To be clear, it's not a full upgrade, it's still based on Wheezy. v0.7 is more of a stop gap while v1.0 hums along slowly.

shanehull, 2017-03-03:
Works great - only Airplay won't appear for me on any devices here. Haven't tested WiFi yet.

kingosticks, 2017-03-03:
Ah OK that's really useful thanks. My only airplay device is elsewhere right now so I didn't try that out. Hopefully that's just a case of recompiling the latest and greatest version of shairport-sync, I'll get on that. 

WiFi for the zeroW and Pi3 was a complete disaster, I've no idea why but it's not recognising the inbuilt wifi device....

shawaj, 2017-03-03:
Any help we can provide?

Michel-J, 2017-03-04:
Tried the 0.7 img, no sound out of the HifiBerry Dac+Pro

kingosticks, 2017-03-09:
@shanehull I really messed up the Airplay support, hopefully that is fixed in v0.7RC2 at https://docs.google.com/uc?id=0ByrmzcQJL5t1aEY0MW42T3VFazg&export=download

@shawaj Testing!

@Michel-J Thank you for testing that, I don't have most of these sound cards so I'm currently totally reliant on help like this. Despite trying my very best not to change necessary things, I couldn't help myself and probably broke it in https://github.com/pimusicbox/pimusicbox/commit/9f7d495a5a192c6c9765a2202b47a287c3faf713. Hopefully I fixed it in https://github.com/pimusicbox/pimusicbox/commit/38664d7c51dc2e2b8b8402a3e929f7857822b3f1. If you are still interested then you can quickly verify that with:
```
sudo wget -q -O /etc/modprobe.d/raspi-blacklist.conf https://raw.githubusercontent.com/pimusicbox/pimusicbox/38664d7c51dc2e2b8b8402a3e929f7857822b3f1/filechanges/etc/modprobe.d/raspi-blacklist.conf
sudo wget -q -O /etc/modules https://raw.githubusercontent.com/pimusicbox/pimusicbox/38664d7c51dc2e2b8b8402a3e929f7857822b3f1/filechanges/etc/modules
sudo wget -q -O /opt/musicbox/setsound.sh https://raw.githubusercontent.com/pimusicbox/pimusicbox/38664d7c51dc2e2b8b8402a3e929f7857822b3f1/filechanges/opt/musicbox/setsound.sh
```

Or try the new v0.7RC2 image at @shawaj Testing!

shawaj, 2017-03-09:
@francesco-vannini will be helping me too :-)

shanehull, 2017-03-10:
Nice one.

I just acquired a HiFiBerry DAC+, so I'll test RC2 with that next week some time!

ateam201, 2017-03-14:
Can't get stopify to work I set it in settings but no music using rc 2 image

shawaj, 2017-03-14:
You definitely have a paid for Spotify premium account?

And you're using your account username not the account email?

ateam201, 2017-03-14:
nah not premium free version i got google music hooked up but the songs show up on the list but again they don't play using speakers hooked up to headphone jack.

shawaj, 2017-03-14:
Only works with Spotify premium unfortunately.

Google music should work though...

kingosticks, 2017-03-16:
As @shawaj said, Premium only.

v0.7.0 RC3 is at https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC3

shawaj, 2017-03-16:
> As @shawaj <https://github.com/shawaj> said, Premium only.
>
> v0.7.0 RC3 is at https://github.com/pimusicbox/
> pimusicbox/releases/tag/v0.7.0RC3

Thanks, think @francesco-vannini has started testing.

Maybe we should set it up in the office next week and use and abuse it....
Could be a good way to find any bugs :-)

What do you think Francesco?

francesco-vannini, 2017-03-16:
Sure, the office is a really good place for stress testing :)

shawaj, 2017-03-16:
Cool let's do it next week!

shawaj, 2017-03-16:
@kingosticks - we (myself and @francesco-vannini ) have been testing today...has support for JustBoom boards been built in at the moment? 

What do we need to get this sorted for the Pi3 release? See #366 also

kingosticks, 2017-03-16:
Not if they need particular drivers (which they probably do). Justboom are
not my favourite supplier as a consequence of their attitude of offering
help and then disappearing once they found other projects, presumably with
bigger bang for their buck. I'm currently not overly interested in
supporting that way of working.

It's probably some very simple changes to setsound.sh (and websettings) to
get it working but I don't have their hardware so it's up to someone who
does.

kingosticks, 2017-03-16:
Hang on!? Do you guys from JustBoom? Do you know who I was emailing, can I get back in contact with whoever I need to and get it working? i.e do you have the hardware to test it?

shawaj, 2017-03-16:
@kingosticks it was probably me or John you were talking to :-)

And really sorry - never intended to leave you hanging whatsoever! We have just been so slammed since the JustBoom launch and because there was only slow progress here and we were not exactly sure what to do (we have also been unfortunately delayed in hiring a full time software dev) we didn't get very far with it. In any case - we want to do everything we can do to help build in support for JustBoom...and happy to provide our support/updates as well as hardware samples to send you of all the boards.

But yes - do you want us to send you all the hardware accross? It is just a case of pulling in the latest Kernel - all the drivers for JustBoom are in there already (justboom-dac and justboom-digi) and they work the same way as the HiFiBerry and IQaudio drivers with the config.txt file (or they auto configure with the onboard EEPROM on the HATs - but the Pi Zero boards do not have EEPROMs).

The only thing we aren't really sure of, is what changes we would need to make in PiMusicBox to get it working

kingosticks, 2017-03-16:
Sounds good, let's take this off into https://github.com/pimusicbox/pimusicbox/issues/366

shawaj, 2017-03-16:
@kingosticks ok see you in #366 - and we will keep testing anyway and will report any bugs back here 👍 

No worries at all either - we did say we could help and I am truly sorry we couldn't have helped more, we have just had a bit of a crazy couple of months and haven't managed to find a suitable software dev to join our team yet so have just been struggling along without one for now....

@francesco-vannini is primed to create guides and do testing on these releases as they come though.

will help in any way we can :-)

iamandiradu, 2017-03-17:
I don`t know if this is v0.7 RC3 related or not, but I can`t acces the browser interface with musicbox.local (in my case: music.local) anymore. It only works with the IP. Is there a quick fix for this, is it my network`s fault or it`s an issue locally on v0.7? Thanks :)

shawaj, 2017-03-17:
Are you on a Mac or PC or other device?

kingosticks, 2017-03-17:
It seems to be a v0.7 issue that avahi/zeroconf/bonjour is broken. Regular dns works if your home router is any good (but I appreciate most are not) I.e http://music/ 

I think it's just a problem with using the old config with the new version but Ive not found the solution to that one yet.

iamandiradu, 2017-03-17:
@shawaj I m using Ubuntu, W10 and Android 6.0. I was able to acces the page using musicbox.local with the 0.6 version but I wanted to use it on my RPi W and it didn t work so I browsed a lil deeper and found this thread with 0.7 which supports W. Then I remembered that I read that 0.7 doesn t work with W s WiFi yet and that I don t have a USB hub so I decided to put it in my RPI 2.

iamandiradu, 2017-03-17:
I installed MusicBox only to listen a radio online (best radio in my country, the others stream :hankey: and ads) and it works. I d love to run it on my W, I m using my RPi 2 for different projects. I m please with MusicBox, keep it up and ,if possible, make it work on W :). Thanks!

kingosticks, 2017-03-17:
@iamandiradu Wifi for Pi3 and Pi0W is fixed in 0.7rc4, see https://github.com/pimusicbox/pimusicbox/issues/393. And .local issue is tracked in https://github.com/pimusicbox/pimusicbox/issues/397 (also fixed in 0.7rc4).

iamandiradu, 2017-03-17:
PERFECT! Thank you :)

iamandiradu, 2017-03-20:
Can you give me a link to grab this 0.7rc4? :)

shawaj, 2017-03-20:
https://github.com/pimusicbox/pimusicbox/releases

iamandiradu, 2017-03-20:
Latest one is RC3 which does not have a fix for RPi 0 W s WiFi :(. kingosticks talked about rc4, but I can t seem to find it.

[6 more replies]

closed 2017-03-26 by kingosticks