pimusicbox/pimusicbox #358: Auto start URL limited u

pengimo, 2016-03-04:


jcass77, 2016-03-05:
This is actually a mopidy-websettings issue and a duplicate of https://github.com/pimusicbox/mopidy-websettings/issues/3.

It was fixed in v 0.1.4.2 - please update to the latest version by running ``pip install -U mopidy-websettings`` and try again.

closed 2016-03-05 by jcass77

kingosticks, 2016-03-05:
That'll update mopidy too, so don't do that unless you know what you're doing.

pengimo, 2016-03-08:
Yup. I ran the command and have broken the box :(

kingosticks, 2016-03-08:
If you can rescue the install (difficult) or are able to reinstall from scratch then next time you can update just the websettings extension with `pip install -I mopidy-websettings==0.1.4.2 `

pengimo, 2016-03-08:
Thanks @kingosticks. I'll reinstall and give that a try.

pengimo, 2016-03-08:
Same result. Performed the commands and the box broke again. I think I'll live with the bug for now and just hardcode the auto URL.

kingosticks, 2016-03-08:
eek sorry. Maybe you must use `-U` instead of `-I`. Basically you don't want to install any of the other dependencies.

pengimo, 2016-03-08:
Both ways failed and left me with a box that didn't want to work. It booted and the network was fine but the web interface was not accessible. Not to worry.