pimusicbox/pimusicbox #360: PC's browser showing saved streams, but mobile browser doesn't.

f3bruary, 2016-03-17:
I've added a few radio streams to musicbox. But although they all show in the web interface on Chromium (arch linux), they don't show in Chrome on Android. On my phone I keep seeing the default streams that were there in the beginning and even if I delete them and clear cache > refresh, I see them again.

kingosticks, 2016-03-17:
If you save them using the browser they are only stored in your browser's local storage and not saved on the server. If you want to have all clients on all machines see the same streams then you need to manually add all your streams to the /boot/config/streamuris.js files. 

This is improved upon in the later version of the webclient which will save the streams to the server within a particular m3u playlist. With this, all clients will see all streams, as soon as they are added. Unfortunately these later webclient versions are not compatible with musicbox 0.6 so it won't be fixed until the next musicbox release. Which is taking a while since my free time has disappeared.

kingosticks, 2016-03-17:
Having reread your original post, I realise I assumed you added the new streams using your browser but that's not actually explicit. How did you add them?

f3bruary, 2016-03-17:
thanks for the quick response. I did indeed add them using the web browser. Bummer that it doesn't save them on the server. I actually found that file in /boot/config, but when I edited it and rebooted my sd card was suddenly corrupted and I had to rewrite the image to disk again.

Looking forward to the next release.

closed 2016-03-17 by f3bruary