pimusicbox/pimusicbox #362: can't download 0.6 image file

DRMH, 2016-03-29:
keeps getting to 145mb and then crashing, any ideas anyone?

jodal, 2016-03-29:
Works for me...

```
$ wget https://dl.mopidy.com/pimusicbox/pimusicbox-0.6.0.zip             
--2016-03-29 13:34:14--  https://dl.mopidy.com/pimusicbox/pimusicbox-0.6.0.zip
Resolving dl.mopidy.com (dl.mopidy.com)... 23.235.43.69
Connecting to dl.mopidy.com (dl.mopidy.com)|23.235.43.69|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 290295899 (277M) [application/zip]
Saving to: ‘pimusicbox-0.6.0.zip’

pimusicbox-0.6.0.zi 100%[=====================>] 276,85M  46,5MB/s   in 6,3s   

2016-03-29 13:34:20 (43,8 MB/s) - ‘pimusicbox-0.6.0.zip’ saved [290295899/290295899]
```

DRMH, 2016-03-29:
OK apologizes maybe it's just me. Tried it again and it didn't work. Turned off everything else on my network that I didn't need and tried again and it did download successfully. Sorry to bother you.

closed 2016-03-29 by kingosticks