pimusicbox/pimusicbox #363: Freshly flashed image on SD card contains no config dir or settings.ini file anywhere

bartkl, 2016-04-01:
I've just downloaded v0.6 and flashed it on my SD card using DD. When I mount it I can browse the filesystem, but there's no config directory anywhere. Performing a recursive search using find gives no results. Also, when searching recursively for '[MusicBox]' using grep, this results in no matches (while this is supposed to be content in the settings.ini file).

Can anyone help me out here? Thanks :)

bartkl, 2016-04-01:
By digging around in some of the install scripts, I stumbled upon `create_musicbox.sh`, which contains the lines:
```
ln -fsn /boot/config/settings.ini /home/mopidy/.config/mopidy/mopidy.conf
mkdir -p /var/lib/mopidy/.config/mopidy
ln -fsn /boot/config/settings.ini /var/lib/mopidy/.config/mopidy/mopidy.conf
```

When mounting the SD card on my laptop, /boot/config does not exist at all, probably because of my mount parameters or something? Anyways, I'll go look for one of the two actual files which are linked to. That would be correct right?

kingosticks, 2016-04-01:
If you mount the SD card on your laptop then then the files should  exist *under the mount point*.

bartkl, 2016-04-01:
Yes, I understand. For clarity: that is indeed where I checked.

Thanks

bartkl, 2016-04-09:
The image contains two partitions, of which one mounted automatically on my system, leaving me unaware of the existence of the other. And that other partition -the first, smaller one- happens to be the one where the `settings.ini` file is located in `/config`.

closed 2016-04-09 by bartkl