pimusicbox/pimusicbox #364: HTTP request to toggle Pause / Play

daviddarke, 2016-04-12:
Is it possible to toggle playback via HTTP request? I recently got hold of a couple of flics https://flic.io/ and there is a HTTP request option out of the box.

I'm assuming that this isn't currently possible?

kingosticks, 2016-04-12:
For Mopidy: yes see http://mopidy.readthedocs.org/en/release-0.19/api/http/

For airplay: I don't know.

daviddarke, 2016-04-12:
@kingosticks Thank you so much for your quick response. So polling curl with:
```
curl -d '{"jsonrpc": "2.0", "id": 1, "method": "core.playback.get_state"}' http://musicbox.local:6680/mopidy/rpc
```
works as expected. Yet the UI for the flic is fairly limited:
[image]
Only allowing for a 'body' field to be sent. I'm assuming they are expecting you to use a REST call.

So I assume this wouldn't work?

kingosticks, 2016-04-12:
Can't you just post `{"jsonrpc": "2.0", "id": 1, "method": "core.playback.get_state"}` as the body?

daviddarke, 2016-04-12:
@kingosticks Seems like a limitation their side: http://ideas.flic.io/forums/283403-flic-ideas/suggestions/7178865-http-calls-post-get-and-rest-api-flic-will-bec Thank you for your help!

closed 2016-04-12 by daviddarke

kingosticks, 2016-04-12:
Indeed it does, hopefully they will improve.