pimusicbox/pimusicbox #365: Using PiMusicBox behind a corporate proxy

alexio102, 2016-04-14:
Fairly new to the coding world, I am trying to setup a PiMusicBox behind a work proxy and I can't find the way to allow the Pi to connect to the internet using that proxy, I've tried editing the mopidy.conf with the config settings on the mopidy website.

[proxy]
scheme = http
hostname = 10.136.205.85
port = 9420

these are correct for my work but these settings do not appear to be working as I still can't access anything from the web.

Can you please help?

fumblingwithit, 2016-04-14:
@alexio102  Do you want to access the internet from your pi in general, or only allow mopidy to access?

bengalaviz, 2016-04-14:
I did this exact thing because I am also behind a corporate proxy.

https://bengalaviz.com/2016/02/11/pi-musicbox-openvpn/

alexio102, 2016-04-15:
I really want only mopidy to access the proxy. I can perform most things by SSH and using the export proxy command but mopidy doesn't seem to use the proxy. I've tried editing the mopidy.conf file but it doesn't help.

jcass77, 2016-04-15:
Not sure if it is relevant, but does the corporate proxy require authentication with a username / password?

I'm not sure that Mopidy supports that.

kingosticks, 2016-04-15:
Most extensions do not support proxied access in the particular versions of the extensions that are used in musicbox v0.6. 

In Mopidy v0.19, I think the config setting will only affect the stream backend (since it's bundled with Mopidy itself rather than an extension).  Did you try to add and play a regular stream?

alexio102, 2016-04-15:
No fortunately it is just a http web address and a port that allows external connection.

alexio102, 2016-04-15:
I did but that would not play either, thinking of just running Jessie and mopidy on the Pi, does the pi with this config run using proxy access?

bengalaviz, 2016-04-15:
Maybe try a program like proxychains. It would force all network traffic through your proxies to connect to the Internet for just the program you want.

kingosticks, 2016-04-15:
> does the pi with this config run using proxy access?

Sorry, I don't understand what you are asking.

alexio102, 2016-04-23:
Sorry for the delay, I've moved to Voluminous now and it seems to work well. Thanks for your help :)