pimusicbox/pimusicbox #366: Add support for BoomBerry audio boards

shawaj, 2016-04-16:
Add support for BoomBerry audio boards. The drivers are supported in kernels 4.1.y and 4.4. y - see files:

sound/soc/bcm/boomberry-dac.c
sound/soc/bcm/boomberry-digi.c
sound/soc/bcm/Kconfig
sound/soc/bcm/Makefile
arch/arm/boot/dts/overlays/boomberry-digi-overlay.dts
arch/arm/boot/dts/overlays/boomberry-dac-overlay.dts
arch/arm/boot/dts/overlays/Makefile
arch/arm/boot/dts/overlays/README
arch/arm/configs/bcm2709_defconfig
arch/arm/configs/bcmrpi_defconfig

If you let me know what needs doing in order to support these, happy to do the work and submit a PR :-)

kingosticks, 2016-05-13:
Sorry for the late reply. I'll include these in the next release.

shawaj, 2016-06-21:
Just FYI, BoomBerry has had to change name to JustBoom due to a legal challenge over the name. The drivers are in the same place though.

Just out of interest, when will you be doing the next build?

shawaj, 2016-09-03:
@kingosticks - perhaps we can have a chat over email? We are keen to put one of our software guys on mopidy/musicbox stuff full time and just wanted discuss a few things?

shawaj, 2017-03-16:
@francesco-vannini FYI here is this issue

shawaj, 2017-03-16:
@kingosticks thanks - that is awesome!

Is there a RC built including this that we can test already? Or will that come in a few days?

kingosticks, 2017-03-16:
I'll try and build a new one tonight but it depends how https://github.com/pimusicbox/pimusicbox/issues/370 comes along which is now a blocker for this (as well as Pi3+  Pi0W wifi support). Might be tomorrow / Saturday.

shawaj, 2017-03-16:
Ok cool. We will wait until next week and get it loaded on our office sound
system for some thorough testing :-)

closed 2017-03-18 by kingosticks

kingosticks, 2017-03-27:
Did you guys get a chance to test anything out? I bought a DAC and after having my eardrums nearly blown off (our bug which I keep forgetting about) it sounds good.

shawaj, 2017-03-27:
@francesco-vannini have you tested this yet?

If not, let's do it tomorrow

francesco-vannini, 2017-03-27:
I was waiting for a link to an image i could use. Have I missed it?

shawaj, 2017-03-27:
Yes it's here - https://github.com/pimusicbox/pimusicbox/releases

RC4 :-)

francesco-vannini, 2017-03-28:
OK, found it and installed it, so far so good!

talhoffer, 2017-04-07:
I can also confirm it works out of the box for a Pi3 with a JustBoom Digi HAT.

kingosticks, 2017-04-07:
thanks everyone, I declare it: working!

shawaj, 2017-04-07:
Awesome - upgrading our office one to this updated software too :-)

commit 590629d1 (2017-03-16): Support boomberry-digi, boomberry-dac and phatdac cards. Fixes #366.
commit 15a85227 (2017-03-16): Support justboom-digi, justboom-dac and phatdac cards. Fixes #366.
commit a4c2fbec (2017-03-16): Support justboom-digi, justboom-dac and phatdac cards. Fixes #366.
commit 370260b5 (2017-03-18): Support justboom-digi, justboom-dac and phatdac cards. Fixes #366.