pimusicbox/pimusicbox #368: Spotify - don´t show anything

tnds82, 2016-04-27:
Please help me please

I put the correct credentials spotify, but don't show anything

Trying to  reach MusicBox
Please wait 

And nothing

Best Regards

lukyluke112, 2016-06-20:
Do you have Premium?

shawaj, 2016-06-22:
You need to use your account username rather than the email address

tnds82, 2016-06-22:
I have premium account

shawaj How to put username rather that the e-mail?

Thanks

shawaj, 2016-06-22:
You should have an email address but you also should have a username or
account name. If you signed up with Facebook they will be the same
otherwise they are probably different.

You should be able to find the username from your Spotify my account section

rickvdl, 2016-06-27:
@shawaj I have exactly the same issues, this is what I see in my mopidy.log
```
2016-06-27 09:09:00,052 - ERROR    Spotify login error: Bad username and/or password
2016-06-27 09:09:00,057 - INFO     Disconnected from Spotify
```

I am 100% that I have the correct credentials filled in (username instead of email). If I copy these exact values and paste them in the Spotify web-login then I can login. I indeed have premium.
This is my `/boot/config/settings.ini`

```
# -----------
# | Spotify |
# -----------
# https://github.com/mopidy/mopidy-spotify
[spotify]
# Supply your PREMIUM account credentials to enable Spotify
enabled = true
username = "username"
password = "password"
# The bitrate, the quality of the music played by Spotify, can be set to 96, 160 (default) or 320
bitrate = 320
cache_dir = ""
```

I have no special characters in my password.
Thanks

lukyluke112, 2016-06-27:
Thats how I do it and it works

shawaj, 2016-06-27:
I do it from the "settings" link in the PiMusicBox dashboard and it works fine.

I am not sure why it would not work if you have done it with the correct credentials!