pimusicbox/pimusicbox #369: Continue what it was doing before starting Airplay

rickvdl, 2016-05-11:
When I'm playing a stream, starting Airplay from my iDevice stops the stream and streams the Airplay audio as it should. The only thing is, when I stop Airplay streaming, the audio stream it was playing before doesn't start automatically. 
Is this a bug? Or is it just not implemented yet? If the latter, could someone point me in the right direction so I could take a look?

Thanks a lot.

kingosticks, 2016-05-11:
It's not implemented yet, see https://github.com/mopidy/mopidy/issues/310#issuecomment-124560242

kingosticks, 2016-05-12:
If you want to get something together using mpdtoys as per @blueyed's suggestion then please do and I'll include it in v1.0

closed 2016-12-11 by rickvdl