pimusicbox/pimusicbox #374: Better spotify browsing support

shawaj, 2016-06-21:
Is it possible to add enhanced support for Spotify so that it makes it easier to navigate/browse by bands/artists/songs etc as the options are fairly limited at present?

Thanks

kingosticks, 2016-06-22:
This issue probably belongs in musicbox-webclient rather than here. could you give some concrete examples of what you'd like to see? The webclient already provides links to navigate to the artist and album pages for every song (via the popup context menus).

shawaj, 2016-06-22:
Shall I move over to there then?

Maybe I am missing something, but in spotify integration you can only browse either playlists, or Lists/Genres.

You can't browse by artist etc and you cant use spotify discover?

Thanks

kingosticks, 2016-06-22:
> Shall I move over to there then?

Yes, please.

kingosticks, 2016-06-22:
But I will just say that the webclient can only present the user with the data made available by mopidy-spotify which in turn is dependent on what libspotify has access to. If I remember correctly, libspotify doesn't know anything about Spotify Discover, My Music etc as these features were all added later.

closed 2016-06-22 by kingosticks

shawaj, 2016-06-22:
Is libspotify still being developed or is that obsolete now? Because having things like Spotify Radio, Spotify Discover, Artists, Albums, related music etc would be extremely useful. i.e. all of the things that spotify has in its own web client

kingosticks, 2016-06-22:
Unfortunately not, Spotify abandoned it years ago. Unfortunately it is
still the only way to stream music on Linux/windows since they have not
bothered to replace it. They have got other web APIs but it's difficult to
integrate them into mopidy-spotify since the authorisation mechanisms they
require are very awkward for a client like Mopidy.

shawaj, 2016-06-22:
What makes it so awkward?

Is it something only Spotify could update?

Thanks,
Aaron

kingosticks, 2016-06-23:
See https://github.com/mopidy/mopidy-spotify/issues/16