pimusicbox/pimusicbox #375: Add ability to configure WiFi from a config file

shawaj, 2016-06-21:
Would be really nice if we could configure WiFi without the need to ever plug in to an ethernet connection by setting the SSID/enctryption method/encrtyption key in a config file somerwhere in boot partition?

Thanks

kingosticks, 2016-06-22:
Isn't this already provided by /boot/config/settings.ini?

shawaj, 2016-06-22:
Is that in Raspbian by default?

Just that in your docs it says this is not possible and that you have to do it from the web-app?

Thanks

kingosticks, 2016-06-22:
> Is that in Raspbian by default?

No 

The docs are very, very out of date.

closed 2016-06-22 by kingosticks

shawaj, 2016-06-22:
Sorry, was being an idiot. It is listed on the docs and I just completely missed it.

Tested it and it works perfectly. 

Sorry for being a muppet!

On a side note @kingosticks have you got an email address I could contact you on by any chance?