pimusicbox/pimusicbox #377: Store the Streams URIs on the RPi instead of locally in the browser (cookies)

SiamKreative, 2016-07-25:
Hi there,

Don't get me wrong, I love the MusicBox! But I think it would be so much better that instead of storing the streams URIs locally in cookies, to store them on the Pi itself. Why? Because if you add custom streams from your computer you won't be able to see them on your Android device.

Is this possible?

kingosticks, 2016-07-25:
Hi. In the current version it's not possible but there is an open issue https://github.com/pimusicbox/pimusicbox/issues/352 to support it for the next version. I have closed the other issue in favour of yours since you've explained why this is worth doing.

SiamKreative, 2016-07-26:
Thanks for the answer. I would be happy to help if you can point in the right direction.

What permanent storage methods are available? How about a local writable JSON file on the SD card?

kingosticks, 2017-03-21:
Fixed in https://github.com/pimusicbox/pimusicbox/commit/6b7015e44f2d107a86c52db92b421b758e7c0a32

closed 2017-03-21 by kingosticks

SiamKreative, 2017-03-21:
Fantastic work, I will give it a try soon :)