pimusicbox/pimusicbox #378: Spotify for Family not supported

knopancakes, 2016-07-30:
I get this error:

2016-07-30 02:36:04,915 - ERROR    Spotify login error: User needs a premium account
2016-07-30 02:36:04,926 - INFO     Disconnected from Spotify

My login credentials are correct, and I have Spotify Premium for Family. which I suspect is the issue. So Premium is supported, but Premium for Family doesn't count as Premium for some reason.

kingosticks, 2016-07-30:
Are you the lead member? I had Spotify family briefly when it was on offer and it worked fine for me, but I did not try to log on with the other accounts. 

But this is something you'd have to take up with Spotify themselves as it is their restriction imposed by their login service and there is nothing we can do. I would be interested to hear their response on this though, doesn't seem very reasonable.

knopancakes, 2016-07-30:
Neither lead nor member, just a bored engineer. 
I just switched to regular Premium and hopefully this will resolve the issue. Won't know until the change takes effect next billing cycle. I will check back in with my results in case anyone else has this issue. Not motivated enough to bring this issue to Spotify's attention just yet.

amboleos, 2016-08-06:
I had just tried my family account and it worked. My account is the lead one

shawaj, 2016-08-21:
Works for me with lead account and non lead account.

Think you need to use your username not your email, that's the issue you're facing I believe

closed 2016-10-03 by jodal