pimusicbox/pimusicbox #379: Pi3 broadcom

lklemo, 2016-08-27:
Hello, any suggestion how to adopt 0.6 image to use the built in wifi on PI3,
even managed to boot up the 3 with some changes, the webservices are not starting since it is not able to recognize the broadcom wifi at all.
tried update and upgrade but the same

any ideas please?

kingosticks, 2016-08-27:
Is this not covered by https://github.com/pimusicbox/pimusicbox/issues/357 ? 

I don't think I've anything to add to that, right now.

lklemo, 2016-08-29:
i have checked #357 did boot and able to use only with lan IF.
trying to follow any of variants claiming to have solved the internal wifi or the exeternal links till now with no success.

thanks

closed 2016-08-29 by lklemo