pimusicbox/pimusicbox #380: Spotify playlists not showing

rseeg, 2016-09-21:
About one week ago, the webclient no longer shows my list of Spotify playlists.  No other changes were made to the pi musicbox (using V0.6).  The playlists are still visible on Spotify.

I have reimaged the SD card but this has not made any difference.

Thx

berrydejager, 2016-09-25:
I found a solution for this. However thus workaround needs to have a permanent fix;

[http://virtual.berrydejager.com/pimusicbox-not-showing-spotify-playlist-fixed/](http://virtual.berrydejager.com/pimusicbox-not-showing-spotify-playlist-fixed/)

rseeg, 2016-09-30:
Thanks for the response.
I managed to fix the issue by removing the cache in the var/cache/mopidy/spotify directory which has been fairly well documented.

closed 2016-09-30 by rseeg