pimusicbox/pimusicbox #383: Volume balance (left/right)

McKlavishnikov, 2016-11-20:
Hi! 
I'd like to control volume balance on my PiMusicbox. 
I've read [here ](https://github.com/pimusicbox/pimusicbox/issues/304) that Mopidy not supporting left/right balance yet, and it's OK, because I can change it in alsamixer. But after reboot volume is resetting (it's happening somewhere in /opt/musicbox/setsound.sh)
So, could you please help, how can I save my volume balance preferences???

McKlavishnikov, 2016-11-21:
Who-ho!!!
Edited /opt/musicbox/setsound.sh
in 245 line change from:
`amixer set -c $CARD "$CTL" 96% unmute > /dev/null 2>&1 || true `
to
`amixer set -c $CARD "$CTL" 96%,80% unmute > /dev/null 2>&1 || true `
(96% to left channel, 80% to right) – works great. Sorry for bothering.

closed 2016-11-21 by McKlavishnikov

kingosticks, 2016-11-21:
Apologies for not getting around to replying to this and thanks for updating with your solution. It's helpful as a reminder for something that needs to go in the docs regarding the default volume config setting (when it gets fixed...).