pimusicbox/pimusicbox #385: Time resets after each reboot

McKlavishnikov, 2016-11-24:
Time resets to 25th April 2015 after each reboot. Is there a way to fix it?
Thx.

kingosticks, 2016-11-24:
I think v0.6 uses `ntpdate` to get the updated time on startup. I'm not entirely sure how it's supposed to work but I don't think you can use `ntpdate` if the `ntp` service is running. So you can try the following to stop the service, update the time, and then start the service again:
```
service ntp stop
ntpdate ntp.ubuntu.com
service ntp start
```

I don't have a 0.6 install handy right now to test this, sorry.

mpcaddy, 2016-11-25:
@kingosticks is correct ntpdate won't change the time if the ntp service is running. The other option is to get a RTC clock module for your pi which has a battery and keeps the time when the pi turns off

McKlavishnikov, 2016-11-28:
Guys, sorry, something strange happened.
I've just make a fresh installation of PiMusicbox 0.60 and now time not resetting. Maybe my previous update/upgrade has broken something, maybe something else.
Anyway, thanks for help and looking forward for a new version of PiMusicbox.

closed 2016-11-28 by McKlavishnikov