pimusicbox/pimusicbox #386: Webinterface crashes when there are playlists with wrong simbols in name

McKlavishnikov, 2016-12-10:
Hi!
After copying on Musicbox a series of playlists prepared on Windows and after reboot webinterface didn't launch (though mopidy service was working and I even managed to connect to it via AirPlay). Detecting wrong playlists and deleting them (+reboot) solve my problem. So, for me my problem is solved, but I think it's a bug you should fix in next release (however, looking forward for a new version of PiMusicbox).

Good luck!

jcass77, 2016-12-11:
Could you perhaps provide an extract of the log file (``mopidy.log``) that shows exactly what type of errors occurred?

It is likely that the symbols are causing Mopidy itself to crash (connecting via Airplay does not prove that Mopidy is up and running - that's a feature that is provided via ``shairport-sync``).

The M3U backend went through a major overhaul in Mopidy 2.0, so you may just be encountering an old issue in Mopidy that's since been fixed.

McKlavishnikov, 2016-12-11:
Maybe I'm looking in a wrong place, but at /var/log/mopidy I can't find a logging for more than 2-3 days from now (this error happend 4-5 days ago and there is no record for this time period).
But if this issue is fixed in new version – it's great!!!

kingosticks, 2017-03-21:
I cannot reproduce this in 0.7.0rc4. If this is not fixed then please reopen.

closed 2017-03-21 by kingosticks