pimusicbox/pimusicbox #388: /home/pi/filechanges/../packages/ does not exist

TakeMeNL, 2016-12-28:
https://github.com/pimusicbox/pimusicbox/blob/release-1.0/create_musicbox.sh#L112
This can never work? Because /home/pi/packages doesn't exist.

What you could do is add
`        mv -- "pimusicbox-${PIMUSICBOX_BRANCH}/packages" "$PIMUSICBOX_FILES/../"`
after this https://github.com/pimusicbox/pimusicbox/blob/release-1.0/create_musicbox.sh#L101

kingosticks, 2017-01-03:
Hi, thanks for taking the time to try out this (very experimental) branch. 

Yes, you are right, it's bugged if you just run the script (rather than cloning). This branch is really just a rough proof of concept for moving to Jessie. It won't last much longer.

closed 2017-01-03 by kingosticks

TakeMeNL, 2017-01-03:
To bad I'm running musicbox with that branch en some tweaks on a RPi3.
So far only problem with it is when searching music mopidy hangs for some reason. But that's nothing todo with musicbox.
And alsamixer gain is set to -90 and needs to be set manually on boot. Also getting a error regarding No volume control found

But again that's not the fault of musicbox.

kingosticks, 2017-01-03:
Yeh I only tested it on an RPi3, turns out it rainbow screens on anything else. 

Are any of your other tweaks useful to incorporate?

Don't we already set the gain to something sensible in setsound.sh? It's been so long I forgot what I changed there.

TakeMeNL, 2017-01-03:
Rainbow screens yes!
But only if I do diet-upgrade I'm using a older kernel now, no problems.

And the setsound.sh is only setting output, not remembering something about gain setting.
Could be wrong.

The tweaks I did where some webclient and config files. Nothing special to incorporate

kingosticks, 2017-01-04:
I was thinking of this https://github.com/pimusicbox/pimusicbox/blob/release-1.0/filechanges/opt/musicbox/setsound.sh#L240

commit ae90c110 (2017-01-03): Copy packages when downloading branch from github.

Fixes #388.

Thanks to TakeMeNL