pimusicbox/pimusicbox #390: Play at Stratup Character length

matschfloss, 2017-01-18:
In the settings where you set a local file or a stream to play at bootup.
You can‘t enter a url which is longer than 43 characters.

For example I think this isn´t enough for even a webradio.
You can fill this line with a text editor, then there isn´t a limit for the characters and urls with more than 43 characters are working fine.

If you paste an url which is longer than 43 characters with an editor in the config file, you can‘t save your settings on the webcontrol because of the error "no more than 43 characters".

In conclusion it would be good to deactivate this query since there is no need to limit the characters.

Best regards

kingosticks, 2017-01-18:
Hi, this is a duplicate of https://github.com/pimusicbox/mopidy-websettings/issues/3

closed 2017-01-18 by kingosticks

kingosticks, 2017-01-18:
You need version 0.1.4.2 of Mopidy-Websettings to get this fix. You can see what version you have with:
`pip freeze | grep -i websettings`

You can upgrade Mopidy-Websettings with:
`pip install -U --no-deps Mopidy-WebSettings`