pimusicbox/pimusicbox #391: Play Spotify Playlists Automatically via Cron

CapnGnarly, 2017-02-21:
I have an RPi 3 setup running Pi MusicBox as a player for a large venue. This is just a background music player, which recently replaced an older CD/USB player to allow us to use Spotify. We have three clients that use the venue and we need to automate the playlist/music change throughout the week. I was thinking of doing this via cron so it could be scheduled specifically.

The basic setup is a single Spotify account that has four playlists on it, one for each client and one for us. On Wednesday afternoons, it changes to client A (Genesis), following their building use, it changes back to our list (Hall). On Friday nights, it changes to client B (CR), then back to Hall after their use. On Sunday night, it changes to client C (Delta), then again back to Hall.

Could I use something like `mpc load URI` in order to accomplish this, or are there any other methods that might work?

kingosticks, 2017-02-21:
Since you've also posted at https://discuss.mopidy.com/t/musicbox-playlist-schedule/1512/5 and had some good replies is it OK to close this here? General usage questions usually get better/more answers there than they do on this bug tracker.

CapnGnarly, 2017-02-21:
Please close here. I have found a solution and posted it to the other forum.

closed 2017-02-21 by kingosticks