pimusicbox/pimusicbox #393: Pi Zero W Onboard Wifi Won't Work

cantrollmyrs, 2017-03-16:
Can't seem to find the onboard wifi for a Pi Zero W.

kingosticks, 2017-03-16:
Unfortunately so. I forgot to add that to the GitHub release notes but have done so now. If anyone can figure it out I'm all ears.

kingosticks, 2017-03-16:
We need a newer kernel which is blocked by https://github.com/pimusicbox/pimusicbox/issues/370

dezeyer, 2017-03-17:
I got Wifi working by moving https://raw.githubusercontent.com/RPi-Distro/firmware-nonfree/master/brcm80211/brcm/brcmfmac43430-sdio.txt and https://raw.githubusercontent.com/RPi-Distro/firmware-nonfree/master/brcm80211/brcm/brcmfmac43430-sdio.bin in /lib/firmware/brcm, then reboot.

cantrollmyrs, 2017-03-17:
@dezeyer how did you move it over?  Can I just open the file via sdcard in my mac and paste the txt content in?

cantrollmyrs, 2017-03-17:
Nevermind, replaced both files, but still having wifi issues on pi zero w.  Same network error.

kingosticks, 2017-03-17:
The Solution is coming, need to rebuild the image today, hold on!

closed 2017-03-17 by kingosticks

kingosticks, 2017-03-21:
https://github.com/pimusicbox/pimusicbox/releases/tag/v0.7.0RC4

commit 4fb34acd (2017-03-17): Use Device Tree Overlays to manage modules. Fixes #370.

Upgraded to kernel 4.4.50. Fixes #393, #366.